arduino/Arduino #4787: New Arduino/Genuino library for OPC released

ildemartinez, 2016-03-31:
Hello

I am proud to release the OPC library for Arduino. It is widly used but I would like to be available using the Arduino IDE library manager. This new release has improved code and bug fixed.

Here is the url of the released 1.7

https://github.com/SoftwareTools4Makers/OPC/releases/tag/1.7

The main project page is at: http://www.st4makers.com/

All the best
ilde

agdl, 2016-04-01:
Done thank you!

closed 2016-04-01 by agdl

ildemartinez, 2016-04-06:
Thank you but the 1.6.9 is not released just 1.6.8 and the community cant download the latest version...how could I fix it?

All the best
ilde

agdl, 2016-04-07:
What do you mean? Library manager is independent from IDE Version. I just placed the 1.6.9 milestone just to count close issues between releases