arduino/Arduino #4796: problem of copying a String with nul byte in the middle

Dvanoska, 2016-04-01:
After uploading my sketch (using Arduino 1.6.8 and Arduino Uno), 

void loop()
{

    String str1 = "ACDEFG";
    String str2 = "HIP";
       
    String str3 = str1+'\0'+str2;   
       
    Serial.println(str1);
    Serial.println(str1.length());     
    Serial.println(str2); 
    Serial.println(str2.length());   
    Serial.println(str3);
    Serial.println(str3.length());    
    Serial.println();
    delay(10000);    
 
}

I’m expecting to get as a result something like this:

ACDEFG
6
HIP
3
ACDEFG0HIP
10
 
But the result is:

ACDEFG
6
HIP
3
ACDEFG `ah
10

I.e. if I understand the problem correctly the String object keeps the right length of str3, but while copying the ‘\0’ it decides the string is over, but it’s not. So in str3  I get right symbols before ‘\0’  and some garbage after.

matthijskooijman, 2016-04-12:
See also #1936, which I believe fixes this (but is probably outdated and needs a rebase, which I'll happily do if anyone intends to merge it).

cousteaulecommandant, 2016-05-06:
Might this be because String::concat(cstr, len) uses `strcpy()` instead of `memcpy()`?  strcpy() is not precisely good at copying char buffers which might contain null bytes...
(Also if memcpy is used instead it wouldn't be necessary to create a 2-byte array and adding a \0 for `String::concat(char)`)

matthijskooijman, 2016-05-06:
@cousteaulecommandant, Yeah, that's exactly one of the things changed by #1936, see https://github.com/arduino/Arduino/pull/1936/commits/c63b861241eed1340ad56c0d05be9b65b90c91cb