arduino/Arduino #4798: "Save as.." doesn't save only in new folder

ivanebernal, 2016-04-01:
When I have many files that are saved in the same folder opened in the same window and I create a new file (in the same window), after saving all the files in a new folder selecting "Save As..", the new file is also saved in the original folder. 
I would expect to have the file saved only in the new folder that I created and not in both the new folder and the original folder.

per1234, 2016-04-01:
Try this:
- Open sketch
- Down Arrow > New Tab > enter name for new file > OK - this creates a new file in the original folder
- Edit the new tab
- File > Save As.... > save to the new folder
- File > Open > open original sketch

You should now see that the edit you made to the new tab was only saved to the new folder, not the original folder, thus proving that "Save As..." only saves to the new folder.

In the future it would be helpful if you would provide a clear set of steps necessary to reproduce the issue you report as well as stating which version of the Arduino IDE you are testing with(always test the hourly build before reporting an issue to make sure the problem hasn't already been solved).

ivanebernal, 2016-04-01:
I see now, the file is created when it's named. I'll make sure to provide step by step to replicate my problem in the future. I'm closing this now. Thanks!!

closed 2016-04-01 by ivanebernal