arduino/Arduino #4799: [Library Manager] Please add this library.

NAzT, 2016-04-01:
http://github.com/cmmakerclub/CMMCEasy

per1234, 2016-04-02:
April fools joke? Then not a very funny one.

NAzT, 2016-04-05:
@per1234  Oh sorry, some commits are missing!

I've pushed them up. and here's the build status:

[image]

Thank you!

per1234, 2016-04-05:
Ok, sorry, I thought you got me there(April 1st is a traditional day for pranks). I noticed there was no category specified in library.properties so started to submit a pull request to add one but kept running into more and more bugs, then I realized the date of the issue report!

agdl, 2016-04-05:
@NAzT Your tag says 0.0.0 but library.properties says 0.0.1 fix this please

NAzT, 2016-04-06:
@agdl  done Thanks! <3

agdl, 2016-04-06:
Done thank you!

closed 2016-04-06 by agdl