arduino/Arduino #4802: [Library Manager] library_index.json parsinh creates messy tree

rlogiacco, 2016-04-02:
I'm parsing the `http://downloads.arduino.cc/libraries/library_index.json` resource in order to create a categorized library tree, but the outcome is now a mess.
I submitted 3 issues in the past week with the aim to help cleanup the tree I was getting out which contained a few misspelled categories, but actually I got it even worse, with the same library now belonging to multiple categories.

I can sort this mess in my own code, but I'm here questioning if this shouldn't be solved once and for all at it's root.

Don't you think such cleaning should occur when creating the aggregated library index?

cmaglie, 2016-04-04:
> Don't you think such cleaning should occur when creating the aggregated library index?

Yes, I agree.

> I got it even worse, with the same library now belonging to multiple categories.

Ok, but how is this supposed to be fixed? 
If we set all the wrong categories to "Uncategorized" then the same library will belong to two categories but on different versions (as happened with the "fixed" library).

Maybe we should not publish the library if the category is wrong?

BTW, even doing so, a library can always switch valid category between releases, so we can always end with different versions of the same library belonging to different categories. And IMHO it seems wrong to deny this possibility...

rlogiacco, 2016-04-04:
I would publish the library only in the category where it's more recent version belongs, so to allow for corrections, still delivering all versions in one category.... Would that fit?

cmaglie, 2016-04-04:
Looks reasonable: let's override the category with the one specified on the latest version, this will still allow the author to change category. I'll work on it.

cmaglie, 2016-04-07:
Ok done, it looks like a very good change, some libraries with missing category now have it and some wrong categories have been auto-corrected.

@rlogiacco thank you very much!

```diff
--- library_index.json.copy	2016-04-07 16:33:46.639556602 +0200
+++ library_index.json	2016-04-07 16:34:31.987628000 +0200
@@ -815,2 +815,3 @@
       "website": "http://firmata.org",
+      "category": "Device Control",
       "architectures": [
@@ -2760,3 +2761,3 @@
       "website": "http://kreatech.verkstad.cc/en/course-literature/education-shield/",
-      "category": "Education",
+      "category": "Other",
       "architectures": [
@@ -2780,3 +2781,3 @@
       "website": "http://kreatech.verkstad.cc/en/course-literature/education-shield/",
-      "category": "Education",
+      "category": "Other",
       "architectures": [
@@ -2800,3 +2801,3 @@
       "website": "http://kreatech.verkstad.cc/en/course-literature/education-shield/",
-      "category": "Education",
+      "category": "Other",
       "architectures": [
@@ -2841,3 +2842,3 @@
       "website": "https://github.com/arduinoverkstad/Svante-Workshop",
-      "category": "Robotics",
+      "category": "Uncategorized",
       "architectures": [
@@ -2861,3 +2862,3 @@
       "website": "https://github.com/arduinoverkstad/Svante-Workshop",
-      "category": "Robotics",
+      "category": "Uncategorized",
       "architectures": [
@@ -7621,2 +7622,3 @@
       "website": "https://github.com/ParsePlatform/Parse-SDK-Arduino",
+      "category": "Uncategorized",
       "architectures": [
@@ -7640,2 +7642,3 @@
       "website": "https://github.com/ParsePlatform/parse-embedded-sdks",
+      "category": "Uncategorized",
       "architectures": [
@@ -7972,2 +7975,3 @@
       "website": "http://www.pjrc.com/teensy/td_libs_OneWire.html",
+      "category": "Communication",
       "architectures": [
@@ -8326,3 +8330,3 @@
       "website": "https://github.com/sparkfun/SparkFun_Color_LCD_Shield_Arduino_Library",
-      "category": "Other",
+      "category": "Display",
       "architectures": [
@@ -9046,2 +9050,3 @@
       "website": "http://github.com/sparkfun/MG2639_Cellular_Shield",
+      "category": "Uncategorized",
       "architectures": [
@@ -9165,3 +9170,3 @@
       "website": "https://github.com/sparkfun/SparkFun_ZX_Distance_and_Gesture_Sensor_Arduino_Library",
-      "category": "Sensor",
+      "category": "Sensors",
       "architectures": [
@@ -9345,2 +9350,3 @@
       "website": "http://github.com/256dpi/arduino-mqtt",
+      "category": "Communication",
       "architectures": [
@@ -9364,2 +9370,3 @@
       "website": "http://github.com/256dpi/arduino-mqtt",
+      "category": "Communication",
       "architectures": [
[... 201 lines cut ...]
+      "category": "Other",
       "architectures": [
@@ -21272,2 +21392,3 @@
       "website": "https://github.com/Links2004/arduinoWebSockets",
+      "category": "Communication",
       "architectures": [
@@ -21290,2 +21411,3 @@
       "website": "https://github.com/Links2004/arduinoWebSockets",
+      "category": "Communication",
       "architectures": [
@@ -21907,3 +22029,3 @@
       "website": "https://git.defproc.co.uk/DefProc/Lewis/",
-      "category": "Communiction",
+      "category": "Communication",
       "architectures": [
@@ -22742,3 +22864,3 @@
       "website": "https://github.com/SodaqMoja/Sodaq_dataflash",
-      "category": "Storage",
+      "category": "Uncategorized",
       "architectures": [
@@ -23696,2 +23818,3 @@
       "website": "https://github.com/orgua/OneWireHub",
+      "category": "Sensors",
       "architectures": [
```

closed 2016-04-07 by cmaglie

rlogiacco, 2016-04-08:
:+1: felice di aver contribuito!