arduino/Arduino #4804: Serial monitor on Mac: no output unless opened after sketch is uploaded

Breidenbach, 2016-04-02:
If the serial monitor is started before the upload there is no output.  In this case it must be closed and then reopened.  This also happens if the Arduino is reset - the monitor must be closed and reopened.

Leonardo
nightly build 1.6.9 Hourly Build 2016/04/02 12:12
OS X El Capitan 10.11.4
MacBook Pro

Test sketch:
#include <SoftwareSerial.h>

void setup()
{
  Serial.begin(9600); 
  Serial.println("starting test");
}
void loop() {
  Serial.println("test line");
  delay (2000);
}

sandeepmistry, 2016-04-04:
@Breidenbach since the Leonardo uses native USB for the Serial port you'll have to do something like this to see the "starting test" line:

```arduino
#include <SoftwareSerial.h> // note: backslashes in this comment to make include name printable

void setup()
{
  Serial.begin(9600);
  while(!Serial); // wait for Serial port to enumerate, need for Native USB based boards only
  Serial.println("starting test");
}
void loop() {
  Serial.println("test line");
  delay (2000);
}
```

This will block until the USB CDC port has been enumerated.

Without the line I added, I still see "test line" printed on my MacBook Pro running OS X 10.11.3 and IDE 1.6.8 if I leave the Serial Monitor open while uploading a sketch.

sandeepmistry, 2016-07-18:
I'm closing this due to lack of feedback, I think this is also expected behaviour.

closed 2016-07-18 by sandeepmistry