arduino/Arduino #4805: Arduino due can't exit from bootloader mode often

sined23, 2016-04-03:
Hello! PC is connected to Arduino Due by Native USB. Often after sketch uploading through Native USB or when I plug usb cable or after OS restarted (power restart), Arduino Due still keep in Bootloader mode and usb communication with PC doesn't work and sketch doesn't work.
After hardware reset of Arduino Due it become normal.

lsusb | grep boot
Bus 001 Device 020: ID 03eb:6124 Atmel Corp. at91sam SAMBA bootloader

I run manually bossac's command to reset Arduino Due but it doesn't help:

~/.arduino15/packages/arduino/tools/bossac/1.6.1-arduino$ ./bossac -p ttyACM0 -R

lsusb | grep boot
Bus 001 Device 020: ID 03eb:6124 Atmel Corp. at91sam SAMBA bootloader

Why can't Bossac reset CPU? Is this bug or need send some additional command to CPU to reset?

BTW: Arduino Due can recieve new sketches but can't be reseted.

lulingar, 2017-02-03:
I confirm this is still an use as of February 2017...

agdl, 2017-02-08:
This issue was moved to arduino/ArduinoCore-sam#22

closed 2017-02-08 by agdl

dav322, 2023-02-23:
send this command to the due before using bossac
stty -F /dev/ttyACM0 1200