arduino/Arduino #4821: Builder supports sketch subfolders, but IDE editor does not

PaulStoffregen, 2016-04-05:
Recent versions of Arduino with the new builder support recursive building, where subfolders are scanned and all files found are compiled.  Perhaps this was meant only for libraries, but it's also being done for sketches.

However, the IDE editor does not support files in subfolders.  If you create a subfolder in a sketch folder, and place code within it, this code will be built as part of the sketch.  But the user can't edit it, or even see any visual indication Arduino is building other code.

This can be problem for examples, written before the new builder, which placed PC-based example code in subfolders.  Because there's no visual indication other code is being compiled, this leads to very confusing errors.

lmihalkovic, 2016-04-15:
This is where other people have also commented about the absence of file management capability in the editor. This is actually an easy thing to add (the first change I made to my own codebase). Done reasonably well it is also a big help for integrating a debugger.

per1234, 2017-07-04:
.ino files were excluded from recursive compilation in https://github.com/arduino/arduino-builder/commit/54e0a8fa96317d608fd4a7d514083283998462dc, which goes a long way towards solving this.

I believe recursive compilation in sketch folders was also limited the `src` folder, which makes it much less likely that a user will inadvertently run into problems by putting source files in a subfolder of their sketch folder.

The downside to always showing the contents of the `src` folder in the IDE as tabs would be that it clutters things up significantly. Often libraries are bundled with a sketch specifically so that it can be provided to the end user as a complete package and they will not need to worry about installing libraries. It's this sort of user who would be the least likely to need to access the contents of the bundled libraries from the IDE and the most likely to be confused by them.

I do think it would be nice for the ability to open those files in the IDE to be a non-default option, which would be sensible to do in combination with https://github.com/arduino/Arduino/issues/3512.

per1234, 2020-09-27:
Please see https://github.com/arduino/Arduino/issues/7609 for additional discussion on this topic.

TCB13, 2024-03-10:
> .ino files were excluded from recursive compilation in [arduino/arduino-builder@54e0a8f](https://github.com/arduino/arduino-builder/commit/54e0a8fa96317d608fd4a7d514083283998462dc), which goes a long way towards solving this.

I don't get it. This has nothing to do with the way compilation is handled. The IDE just needs to do what VSCode does, show a sidebar with all the folders bellow the folder where the root .ino file sits and that's it.

per1234, 2024-03-10:
>The IDE just needs to do what VSCode does, show a sidebar with all the folders bellow the folder where the root .ino file sits and that's it.

This is now done in Arduino IDE 2.x:

1. Select **File > Preferences...** (or **Arduino IDE > Settings...** for macOS users) from the Arduino IDE menus.
   The "**Preferences**" dialog will open.
1. Check the box next to "**Show files inside Sketches**".
1. Click the "**OK**" button.
   The "**Preferences**" dialog will close.
1. Click the icon that looks like a folder in the activity bar on the left side of the Arduino IDE window.
   The "**SKETCHBOOK**" view will open.
1. You will see the contents of the sketch folder. Click the **❯** icon to expand the `src` subfolder.
1. Click on the file under the `src` subfolder that you would like to edit.

The file will now open in an Arduino IDE editor tab.

per1234, 2024-03-10:
Since the recursive compilation is now controlled and the IDE allows the user to see and edit the contents of the `src` subfolder, I will close this as resolved.

closed 2024-03-10 by per1234