arduino/Arduino #4823: Dead Link on Arduino Playground Website

ChaitanyaBaweja, 2016-04-05:
http://playground.arduino.cc/Code/SLCD
Unable to access the SCLd library..
Error displayed : Http/1.1 Service Unavailable

per1234, 2016-04-05:
The Arduino Playground content is the responsibility of the community, not the Arduino developers. It looks like you have made a post on the Arduino Forum regarding this issue: http://forum.arduino.cc/index.php?topic=391663. Since it is off topic here I think it would be best if you close this issue and we will continue discussion of how to fix this on the forum.

ChaitanyaBaweja, 2016-04-06:
Ohh. Actually it wasn't me who started that thread in the forum and I
understand your point regarding the off topic issue.. Sorry I thought all
bugs should be reported there, but yes I will surely delete the thread..

per1234, 2016-04-06:
>Actually it wasn't me who started that thread in the forum

That's funny, I just assumed it to be you. I guess there is some interest in this library. I'd appreciate your input on the solutions I have proposed in my comment on the forum thread: http://forum.arduino.cc/index.php?topic=391663.msg2697798#msg2697798  I have posted a link to where you can find the SLCD library in that comment. Per

facchinm, 2016-04-11:
Moving the conversation to the forum, thakns :wink:

closed 2016-04-11 by facchinm