arduino/Arduino #4826: Add the next line will stop any TX data of your arduino

TOTON95, 2016-04-06:
Adding "int main = 0;" before setup() deny any TX Data from your arduino, could anybody confirm it please or if you can explain it :D

TOTON95, 2016-04-06:
** Placing it outside of Setup() or Loop() causes it otherwise it doesnt affect

NicoHood, 2016-04-06:
adding it before setup() is the same as placing it outside setup(). So Your description does not really make sense at all.

Also main is reserved for the main loop hidden by the arduino IDE. Just dont use main as its is a reserved name.

facchinm, 2016-07-19:
Indeed the problem here is about the compiler not emitting any error/warning.
This happens because `main()` is not declared in any header file, thus the compiler legally compile the sketch and the linker replaces main() implementation with an int.
Adding
```
int main( void );
```
to `Arduino.h` should solve the issue.
@cmaglie @sandeepmistry @matthijskooijman , what do you think about that?

cmaglie, 2016-07-19:
@facchinm 
looks fine, I can't see any drawbacks.

TOTON95, 2016-07-20:
@facchinm I can see that you reach  the issue as i did, and must have a warning at least to prevent it. I fixed it changing the name of the variable. 

------------------------------------------------------------------------------------------------------------------
If you consider that this thread has become irrelevant besides the possible addition of a warning to the compiler, let me know to close it, thank you very much for your answers!

facchinm, 2016-07-20:
Let's keep this issue as reference for being closed by the upcoming PR :smile: