arduino/Arduino #4832: beginner need help

xavierartot, 2016-04-06:
Hi,
I add a led and is flashing by itself?
then I wrote this:
```
void setup() {
  // put your setup code here, to run once:
  pinMode(13,OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(13,HIGH);
  delay(5000);
}
```

my config is : Arduino: 1.6.6 (Mac OS X), Board: "Arduino/Genuino Uno"

when I'm downlowd I have those error:

```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware 

"/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/xavi/Library/Arduino15/packages" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools 

"/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -tools 
"/Users/xavi/Library/Arduino15/packages" -built-in-libraries 

"/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/xavi/Documents/Arduino/libraries" -fqbn=arduino:avr:uno -vid-pid=0X2A03_0X0043 -ide-version=10606 -build-path 
"/var/folders/t1/dzcbnfkj0x926nw4zcrn9_t80000gn/T/buildbda9d1b1fc46b7d3fdb544edfa300ef3.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/Users/xavi/Documents/Arduino/helloworld/helloworld.ino"
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/xavi/Library/Arduino15/packages" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/xavi/Library/Arduino15/packages" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/xavi/Documents/Arduino/libraries" -fqbn=arduino:avr:uno -vid-pid=0X2A03_0X0043 -ide-version=10606 -build-path "/var/folders/t1/dzcbnfkj0x926nw4zcrn9_t80000gn/T/buildbda9d1b1fc46b7d3fdb544edfa300ef3.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/Users/xavi/Documents/Arduino/helloworld/helloworld.ino"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/xavi/Library/Arduino15/packages/arduino/hardware/avr/1.6.10/cores/arduino" "-I/Users/xavi/Library/Arduino15/packages/arduino/hardware/avr/1.6.10/variants/standard" "/var/folders/t1/dzcbnfkj0x926nw4zcrn9_t80000gn/T/buildbda9d1b1fc46b7d3fdb544edfa300ef3.tmp/sketch/helloworld.ino.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/xavi/Library/Arduino15/packages/arduino/hardware/avr/1.6.10/cores/arduino" "-I/Users/xavi/Library/Arduino15/packages/arduino/hardware/avr/1.6.10/variants/standard" "/var/folders/t1/dzcbnfkj0x926nw4zcrn9_t80000gn/T/buildbda9d1b1fc46b7d3fdb544edfa300ef3.tmp/sketch/helloworld.ino.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/xavi/Library/Arduino15/packages/arduino/hardware/avr/1.6.10/cores/arduino" "-I/Users/xavi/Library/Arduino15/packages/arduino/hardware/avr/1.6.10/variants/standard" "/var/folders/t1/dzcbnfkj0x926nw4zcrn9_t80000gn/T/buildbda9d1b1fc46b7d3fdb544edfa300ef3.tmp/sketch/helloworld.ino.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/xavi/Library/Arduino15/packages/arduino/hardware/avr/1.6.10/cores/arduino" "-I/Users/xavi/Library/Arduino15/packages/arduino/hardware/avr/1.6.10/variants/standard" "/var/folders/t1/dzcbnfkj0x926nw4zcrn9_t80000gn/T/buildbda9d1b1fc46b7d3fdb544edfa300ef3.tmp/sketch/helloworld.ino.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/xavi/Library/Arduino15/packages/arduino/hardware/avr/1.6.10/cores/arduino" "-I/Users/xavi/Library/Arduino15/packages/arduino/hardware/avr/1.6.10/variants/standard" "/var/folders/t1/dzcbnfkj0x926nw4zcrn9_t80000gn/T/buildbda9d1b1fc46b7d3fdb544edfa300ef3.tmp/sketch/helloworld.ino.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/xavi/Library/Arduino15/packages/arduino/hardware/avr/1.6.10/cores/arduino" "-I/Users/xavi/Library/Arduino15/packages/arduino/hardware/avr/1.6.10/variants/standard" "/var/folders/t1/dzcbnfkj0x926nw4zcrn9_t80000gn/T/buildbda9d1b1fc46b7d3fdb544edfa300ef3.tmp/sketch/helloworld.ino.cpp" -o ""
avr-g++: error: missing filename after '-o'
exit status 1
Error compiling.```

per1234, 2016-04-07:
I have provided advice on how you can fix this issue on your Arduino Forum post: http://forum.arduino.cc/index.php?topic=392056. If you have any further questions you can ask them on that thread.

xavierartot, 2016-04-07:
thanks for that

closed 2016-04-07 by xavierartot