arduino/Arduino #4835: cant burn bootloader

cjdg, 2016-04-08:
I bought a couple of ATMEGA328P-PU and wanted to burn the bootloader on them, i took the old ATMEGA from the UNO and put them on the board and then try to burn the bootloader no sucess, here is what i get:

Arduino:1.6.8 (Linux), Tarjeta:"Arduino/Genuino Uno"

/opt/arduino-1.6.8/hardware/tools/avr/bin/avrdude -C/opt/arduino-1.6.8/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -cstk500v2 -Pusb -e -Ulock:w:0x3F:m -Uefuse:w:0x05:m -Uhfuse:w:0xDE:m -Ulfuse:w:0xFF:m 

avrdude: Version 6.0.1, compiled on Apr 14 2015 at 19:04:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/opt/arduino-1.6.8/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/cjdg/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : stk500v2
avrdude: usbdev_open(): did not find any USB device "usb"
Error quemando bootloader

i also tried this:
/opt/arduino-1.6.8/hardware/tools/avr/bin$ /opt/arduino-1.6.8/hardware/tools/avr/bin/avrdude -C/opt/arduino-1.6.8/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -carduino -P/dev/ttyACM0 -e -Ulock:w:0x3F:m -Uefuse:w:0x05:m -Uhfuse:w:0xDE:m -Ulfuse:w:0xFF:m 

and still get this:
vrdude: Version 6.0.1, compiled on Apr 14 2015 at 19:04:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/opt/arduino-1.6.8/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/cjdg/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM0
         Using Programmer              : arduino
ioctl("TIOCMSET"): Broken pipe
ioctl("TIOCMSET"): Broken pipe
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00

mikaelpatel, 2016-04-08:
What ISP and tutorial are you following?

agdl, 2016-04-12:
@mikaelpatel if I understood well the problem is that @cjdg is trying to burn the bootloader without a programmer. That's the point. You can't burn bootloader without an external programmer (i.e Arduino as ISP)

cjdg, 2016-04-13:
i did it at the end, made a external programmer and attached to the arduino as a shield, during this, i started to think some new issues, first why we cant burn the bootloader directly on the arduino? and the second one, with the atmega finally programmed with the bootloader, when i try to burn the program to the atmega, actually both atmegas the one on the arduino and the one on the shield got burn with the same program, even choosing arduino as isp, any idea?

agdl, 2016-06-06:
@cjdg the bootloader is a piece of firmware that allows the micro-controller to let's say autoprogram it self so on a brand new micro this is missing and you need to burn it. About the second question, I suggest you to move your concerns on forum.arduino.cc
Github is intended to track bugs and not ot give assistance.

Thank you for cooperation

closed 2016-06-06 by agdl