arduino/Arduino #4837: WString.h is not compatible to c++11 range-based loop

Rotzbua, 2016-04-08:
I tried to use the new c++11 syntax with the default string class from arduino. Actually it fails because it has no `begin` or `end`.

Test-IDE: 1.6.8 (Win8.1 x64)
Board: Uno

Example code:
```arduino
String temp = "abc";
for (auto tempchar : temp) {
    Serial.println(tempchar);
}
```

Error:
```
error: 'begin' was not declared in this scope
for (auto tempchar : temp) {
error: 'end' was not declared in this scope
```
Reference:
http://en.cppreference.com/w/cpp/language/range-for

Best

NicoHood, 2016-04-08:
This was already "discussed" somewhere else. No answer as I remember. I also suggest to add this feature.

Chris--A, 2016-04-09:
I have a PR hanging around: https://github.com/arduino/Arduino/pull/2179

I proposed this before C++11 was enabled.

cmaglie, 2016-04-20:
Fixed with #2179

closed 2016-04-20 by cmaglie