arduino/Arduino #4839: Arduino 1.6.8 always reset boards

wwwusr, 2016-04-08:
Hello guys,
Use Arduino1.6.8 in Win7 64,I plug 4 Arduino boards:2 Pro mini(through PL2303),1 Nano,1 Uno(all uses  CH340G),when I open the 1.6.8 IDE,all board's 13 LED will flash,means that the boards are all reset time an time again,and sometimes can upload codes,but sometimes can't do that.
And,1.0.5 IDE version is OK, after run 1.0.5,all boards  run stable,even run some codes throuth serial port.

wwwusr, 2016-04-10:
Version 1.5.8 is OK,tested

PaulStoffregen, 2016-04-10:
Any chance you could test with a genuine Arduino Uno, which uses the 16u2 chip (not ch340).

facchinm, 2016-04-11:
Closed by merging https://github.com/arduino/Arduino/pull/4792, try the latest nightly build and the problem should disappear

closed 2016-04-11 by facchinm

wwwusr, 2016-04-11:
Thanks!I'll try like that