arduino/Arduino #4841: [Library Manager] Please add 6-digit-7-Segment-Arduino

SohnyBohny, 2016-04-09:
Repo is here:
https://github.com/SohnyBohny/6-digit-7-Segment-Arduino

Most recent tag is here:
https://github.com/SohnyBohny/6-digit-7-Segment-Arduino/releases/tag/v1.0

Thank you!

agdl, 2016-04-11:
Done thank you!

closed 2016-04-11 by agdl

SohnyBohny, 2016-04-11:
Installing throws an error 

The library "6-digit-7-Segment" cannot be used.
Library names must contain only basic letters and numbers

I've renamed to "six-digit-seven-segment" in my library.properties

SohnyBohny, 2016-04-15:
@agdl please help! - what should I do?

agdl, 2016-04-15:
@SohnyBohny you have to tag the library again. Make a new release!

SohnyBohny, 2016-04-15:
done - thank you!

SohnyBohny, 2016-04-15:
done - thank you @agdl for your great help!

agdl, 2016-04-15:
You are welcome. In within one hour you should see it in the library manager

SohnyBohny, 2016-04-15:
@agdl would the old one be deleted if I point it as a Pre-release?

agdl, 2016-04-15:
No you have to manually remove the tag from git and then the library manager will remove it as well!

SohnyBohny, 2016-04-15:
Ok - and what's about the colors

agdl, 2016-04-18:
What do you mean with the colors?

SohnyBohny, 2016-04-18:
I've added a `keywords.txt` file for highlighted in 

>  see:
[If you tried the new sketch, you probably noticed that nothing from our library was recognized by the environment and highlighted in color. Unfortunately, the Arduino software can't automatically figure out what you've define in your library (though it would be a nice feature to have), so you have to give it a little help. To do this, create a file called keywords.txt in the Morse directory. It should look like this:](https://www.arduino.cc/en/Hacking/LibraryTutorial)

What did I wrong?

agdl, 2016-04-19:
Try putting one tab \t between the word you want to highlight and the KEYWORDx keyword