arduino/Arduino #4843: OSX Function not declared in this scope error message

rickspratt, 2016-04-10:
The following code produces the error "'HSBtoRGB' was not declared in this scope".  This appears to only happen in a OS X environment. This error occurs in 1.6.7, haven't tested it any other version of the Arduino IDE.  The code will work if you remove the newline from the parameter list in the Function declaration.

int r,g,b,hue,brightness,saturation;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
hue = 180;
brightness = 255;
saturation = 55;

HSBtoRGB(hue, saturation, brightness, &r, &g, &b);

    Serial.print(" hue=");
    Serial.print(hue);
    Serial.print(" saturation=");
    Serial.print(saturation);
    Serial.print(" brightness=");
    Serial.print(brightness);
    Serial.print(" red=");
    Serial.print(r);
    Serial.print(" green=");
    Serial.print(g);
    Serial.print(" blue=");
    Serial.print(b);
    Serial.println("");
    delay(500);
    
}

void HSBtoRGB(int inHue, int inSaturation, 
int inBrightness, int *oR, int *oG, int *oB ) {

  *oR = inHue;
  *oG = inSaturation;
  *oB = inBrightness;
}

per1234, 2016-04-11:
Please test using the hourly build http://www.arduino.cc/en/Main/Software#hourly. This appears to be a duplicate of https://github.com/arduino/arduino-builder/issues/80 which was fixed by https://github.com/arduino/arduino-builder/commit/fbfaa5603aa984b88d2a295658a1a57d57415b3a.

I get the same error with Arduino IDE 1.6.7 and 1.6.8 on Windows 7 so I don't think it's OSX specific. The code compiles without error for me using the hourly build.

facchinm, 2016-04-11:
Thanks @per1234 , the issue with multiline declaration has been solved in mainline and will be available with IDE 1.6.9

closed 2016-04-11 by facchinm