arduino/Arduino #4845: folded code gets eaten by search and replace action

MOOSFETT, 2016-04-10:
hello,

if you have folded some code in a sketch, the IDE eats unpredictable amounts of the code.

you can see for yourself by executing the following steps:
- freshly open arduino IDE
- create a new sketch (file->new or Datei->Neu or STRG-N)
- fold both the setup and the loop routine code
- use the replace-function and replace a word of your favour, e.g. replace "code" with "cidre"
- it doesnt matter wheather you use case sensitive or insensitive search

first you will notice, that the IDE does some computing, and then the folded "void loop() {" disappears.

second - if you save the file in this condition, the eaten code is finally gone.

third - if you try UnDo and ReDo (before saving), some unpredictable behaviour appears on the code, furthermore you are not able to restore the original code as it existed before search and replace action.

and last but not least the search and replace fails with replacing the word "code" which is located in the disabled ( /* ) second line (unfolded condition).

this bug is a bit of subtle and did cost me hours of searching errors, caused by literally an erosion of code... 
furthermore, i think the bug has a history of some arduino-versions, because my first irritations about eroded code date back to late 2014 when my projects grew larger so that i began to use code folding.

i am using a german language arduino 1.6.8 on a windows 7 64bit PC, 16 GB RAM, core-i5 3570K.
the user-account-type is local administrator, no domain.
the arduino-folder is at the default location.

jsotola, 2016-05-23:
i observed same behaviour win7 x64
arduino 1.6.10 hourly build 2018/05/18 07:34

facchinm, 2016-07-19:
Nasty nasty bug, I never experienced it since I don't use code folding but it's there and it's ugly.
@matthijskooijman , do you have any idea? Maybe it could be solved by https://github.com/arduino/Arduino/pull/4363 ?

matthijskooijman, 2016-07-19:
@facchinm, I can't think of the cause for this bug off-hand, but it might indeed be fixed by that PR. I'll make sure to keep this one in mind when continuing work on the editor.

lmihalkovic, 2016-07-19:
Last i knew, it did not. Pb is the search is very _inadequate_ code.

per1234, 2016-08-01:
If this isn't going to be fixed by the next release the code folding option should be removed. The feature is already practically useless for most sketches large enough to need it due to https://github.com/arduino/Arduino/issues/3441.

closed 2016-08-26 by facchinm