arduino/Arduino #4846: Library manager scraper did not include the latest tag of our library

iSsO, 2016-04-11:
It has been more than 14 hours since we tagged the latest version of our library and it hasn't been included in the library manager yet.

Here is the link of the tag: https://github.com/Integreight/1Sheeld-Arduino-Library/releases/tag/v1.9.0

It is mentioned in the [FAQ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ) that new tags should be available within 1 hour. Should I be worried?

Is there a way we can track the status of the library manager scraper? (i.e. a [status page](http://statuspage.io) for instance)

facchinm, 2016-04-11:
Hi @iSsO , we are experiencing some problems on our Jenkins machine, the libraries will be updated as soon as possible

iSsO, 2016-04-11:
@facchinm Thanks, it has been updated.

I suggest you add a status page as I mentioned to avoid this kind of posts. :)

closed 2016-04-11 by iSsO