arduino/Arduino #4857: Serial port not listed on OS X after solve_serial_windows branch was merged

igrr, 2016-04-13:
I am using a CP2102-based USB-serial adapter on OS X 10.11.3. The adapter is present at `/dev/tty.SLAB_USBtoUART` and `/dev/cu.SLAB_USBtoUART`.

Before "solve_serial_windows" branch was merged (i.e. at commit 4c901d3f), I could see my serial adapter in the ports list. After the merge, it is not present in the list (although some other ports are still present).

Running `build/macosx/liblistSerials-1.0.5/osx/listSerialC` gives:
```
/dev/cu.SLAB_USBtoUART | 10C4:EA60 | 0001 CP210x USB to UART Bridge Controller Silicon Labs
```

In the new version (`build/macosx/liblistSerials-1.1.0/osx/listSerialC`) i get:
```
/dev/cu.SLAB_USBtoUART | 10C4:EA60 | (null) (null) (null)
```

Anything I can do to help fix this issue?
Thanks.

facchinm, 2016-04-13:
Hi Ivan,
the new version splits the discovery into two phases, one "light" (which only recovers vid and pid) and another "heavy" which also gets the serial number directly from the device (making an USB transaction).
The `listSerialC` utility now uses the light function so extra info are not displayed, but the port should appear in the IDE as it did before.
Is the board a nodeMCU or something similar? I'm checking with OSX on it and report here the results

igrr, 2016-04-13:
Yep, it's a NodeMCU 1.0 board. 
Thanks for looking into this.

facchinm, 2016-04-13:
I don't have a NodeMCU 1.0 at the moment (I'll be able to test it tomorrow) but with 0.9 (which mounts a CH340G) everything works correctly on El Capitan.
I'll keep you posted as soon as I can test the 1.0

igrr, 2016-04-13:
Just tested with CH340G-based NodeMCU 0.9, it is indeed listed when running the latest hourly build.
FT232 is also listed, unsurprisingly. Also I noticed that `/dev/cu.Bluetooth-Incoming-Port` is listed, although it doesn't appear in `listSerialC` output. Still no dice with CP2102 (I'm using Silabs-provided driver v4.10.7, in case that matters).

facchinm, 2016-04-15:
Hi Igor, the problem was caused by the name of the port (it had an underscore in it) so it was dropped. Fixed by 2ddbf01, it will be available in nightly soon!

closed 2016-04-15 by facchinm