arduino/Arduino #4858: Library conflict-GSM-Keypad, still around

DouglasBair, 2016-04-13:
Take a blank Arduino sketch, include <GSM.h> and <Keypad.h>. try to compile.  I receive this error:
"C:\Program Files (x86)\Arduino\libraries\GSM\src/GSM3MobileAccessProvider.h:37:36: note: previous declaration 'GSM3_NetworkStatus_t IDLE'
 enum GSM3_NetworkStatus_t { ERROR, IDLE, CONNECTING, GSM_READY, GPRS_READY, TRANSPARENT_CONNECTED, OFF};"

Researching this, I see many years of users asking that this be fixed, and not expecting to be told 
fix it yourself by editing a library.  What then for the next release, fix it again? 
I've already check the forum and the issues.  Many 'smart alecs' say its fixed, where?
This happens in v1.6.5, 1.6.6, and in (.org's) 1.7.9.

When will Arduino.cc fix this?  
And, yes, I'm a newbie. So if I'm expecting too much or have entered the lib names incorrectly,
let me know.
Doug

Chris--A, 2016-04-24:
I have written a solution to the problem here (not Arduino specific): https://github.com/Chris--A/Keypad/issues/4#issuecomment-213909336

I will update the Arduino.cc playground with a link to the repo wiki documentation for others (once I have created it).

This issue can be closed.

DouglasBair, 2016-04-24:
Thank you.
Doug

  _____  

be fixed! (#4858)

Chris--A/Keypad#4
(comment)

<https://github.com/arduino/Arduino/issues/4858#issuecomment-213909514> it
on GitHub
<https://github.com/notifications/beacon/ARmIyFq7KLyCm9P6f9eMjAZjDhsXaOz0ks5
p6yMIgaJpZM4IG0xa.gif> 

  _____  

No virus found in this message.
Checked by AVG - www.avg.com
Version: 2016.0.7539 / Virus Database: 4556/12091 - Release Date: 04/24/16

agdl, 2016-07-07:
@Chris--A I'm closing this. Thank you

closed 2016-07-07 by agdl

DouglasBair, 2016-07-07:
How has this been fixed?
 
Is there a new GSM library that doesn't conflict with the Arduino Keypad
lib?
 
Douglas Bair
 
  _____  

be fixed! (#4858)

<https://github.com/arduino/Arduino/issues/4858#issuecomment-231095044> it
on GitHub, or mute
3bks5qTQ2ggaJpZM4IG0xa> the thread.
<https://github.com/notifications/beacon/ARmIyDzrWCXvnKKigkn3OHP0C1wbFq2Pks5
qTQ2ggaJpZM4IG0xa.gif> 

  _____  

Version: 2016.0.7640 / Virus Database: 4627/12571 - Release Date: 07/07/16