arduino/Arduino #4859: Display numeric values in the serial plotter?

damellis, 2016-04-13:
It might be nice to display numeric values in the serial plotter, e.g. next to the corresponding color square in the upper right corner.

madsdyd, 2020-04-07:
There is a workaround for this, that I believe is usable.

Use output like this:
```
Serial.print("RAW:");Serial.print(distance); Serial.print(" "); Serial.print(distance); Serial.print(":"); Serial.print(" ");Serial.println("");
```

This will produce data like this:

`RAW:123.4 123.4:`

which will be interprented as two data series, where the seconds data values are ignored. The first data series will be shown with the label RAW, the second will have its data (which is missing) still be showed in the label (the color square), using the current value as a label.

Of couse, this only shows the last value in the data series, but that should fit your description.

It is not pretty, but it is sort of a edge wish anyway, IMHO.

@per1234 : Recommend close as "wont fix /workaround exists"

per1234, 2022-11-18:
Resolved by addition of point values shown on hover in Arduino IDE 2.x Serial Plotter:

[image]

closed 2022-11-18 by per1234