arduino/Arduino #4862: Menu option to "Clean" the Build folder

steronydh, 2016-04-14:
When using a specified Build folder, if you have two sketches open at once, the Build folder isn't cleaned out between compiles, which can lead to spurious "already defined" compiler errors. To solve this, it would be useful to have a menu item which cleans out the Build folder.

facchinm, 2016-04-14:
@steronydh , are you experiencing this with a nighlty build? Since https://github.com/arduino/arduino-builder/commit/47f3fc4bf51816b7874230548d36db30ce2e594f was merged, I never got that error (since the .o generated by the previous sketch is ignored)
Could you provide a couple of sketches which trigger the issue?

steronydh, 2016-04-14:
Yes, it's a nightly. I'm getting this with two very similar sketches which access the same set of libraries. I think it could be the libraries which are causing this.

facchinm, 2016-04-14:
Sorry but I need the sketches and the environment (core version, target boards etc), otherwise I'll never be able to replicate it coherently

steronydh, 2016-04-14:
With nightly 1.6.9 2016/04/07 12:33, I can reproduce this in this way compiling for Mega2560. 

Specify a build folder in preferences.txt: build.path=C:\Temp\Arduino\Build

Clean out the build folder.

Open two sketches. At the top of each add the global line: <int i=0;>

Compile one. No errors.

Compile the second. I get errors about multiple definitions of setup(), loop(), and i.

facchinm, 2016-04-14:
> Specify a build folder in preferences.txt: build.path=C:\Temp\Arduino\Build

This is the key, since usually two different sketches get compiled into different folders. I'll look into this but don't expect a fast solution (since the option is not reachable by the GUI)

steronydh, 2016-04-14:
OK, no worries. Now there's the option to export the compiled binary to the sketch folder, I can probably delete this setting anyway.

steronydh, 2016-04-14:
I can confirm that the problem goes away if I delete the line completely from preferences.txt, but not if I leave `build.path=` blank.

steronydh, 2016-04-14:
There is a problem with my original build now - gcc complains that the command line is too long! If I use my specified path it's fine, but the random path makes the linker call *very* long. (My project is quite big, producing a 100k+ hex file.)

matthijskooijman, 2016-04-16:
@facchinm, I think the changes I proposed in https://github.com/arduino/arduino-builder/issues/86 would fix this (the duplicate symbol issue, not the linker commandline length). However, it would then still recompile *everything* whenever switching  between Windows, but at least it would work.

@steronydh, perhaps you could set a different build.path for each window / sketch by using the `--pref build.path=/some/path` commandline option? E.g. by creating two shortcuts or wrapper scripts with different options allows setting static, but different, build paths.

cmaglie, 2016-05-06:
Fixed by https://github.com/arduino/arduino-builder/pull/144

closed 2016-05-06 by cmaglie

piotrjan-kano, 2019-04-07:
clean the temp folder where arduino saves temp files. Restart the IDE. Arduino will rebuild