arduino/Arduino #4863: Multiple equal flash-strings are not optimized

Ivan-Perez, 2016-04-14:
Hello!

After looking at my growing-size sketch, which contains too many string messages, I've realised that if a flash string appears more than once the compiler is not saving space and instead it is allocating flash memory for every repeated flash string.

Here it is a simple test case:

```cpp
void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
  Serial.print("A very long example string... A very long example string... A very long example string... A very long example string...");
}
```

Compiled using Arduino Mega as target, these are the sketch sizes and global variables usages:
* All strings: 2990 / 306
* Only one string (commenting all prints out except one): 2864 / 306

When strings are stored in RAM the compiler detects correctly that all the strings are the same, so it can optimize them to save only one of them.

But if I convert all strings to flash-strings (using `F()`), it seems that every string is copied onto the sketch, without optimizing them:
* All strings: 4814 / 186
* One string: 2864 / 186

Could this be improved?

darrylp1per, 2016-05-24:
http://forum.arduino.cc/index.php?topic=194603.0

has a fully working solution. trouble is,  its fixed in a file, not normally altered by end users.

Darryl

matthijskooijman, 2016-05-24:
@darrylp1per, thanks for pointing that out, I wasn't aware of that approach yet. However, looking through the forum thread, it seems there are still some unsolved corner cases with that solution, so I do not think merging it into the Arduino core as-is is a good idea.

On a related note, I believe avr-gcc version 5 has overhauled the PROGMEM approach, using the concept of memory spaces instead of using a variable attribute. I'm not sure if that will actually fix this particular issue, but it does mean that any PROGMEM-based solutions might not be very future-compatible.

darrylp1per, 2016-05-24:
I quite agree,   its a GCC issue to be fixed really.
I'm personally still using GCC 4.8.2  but the 4.9.2 and 5.2.1 versions I also have installed have issues that for me, means I stay on an older version.  

the link I posted will allow Ivan to get his program working I'm 99% sure.  But GCC needs fixing for this optimisation to work properly and without altering the file in $(INSTALL_PATH)\tools\avr\avr\include\avr\pgmspace.h where the PSTR macro normally lives.