arduino/Arduino #4868: fast serialUSB read

sined23, 2016-04-15:
Hi guys! I increased serialUSB read speed from 1Mbit/s up to 12Mbit/s.
Detail are here http://forum.arduino.cc/index.php?topic=391757.0
I suggest you to take it for new arduino version.

facchinm, 2016-04-15:
Hi @sined23 ,
would you mind opening a proper PR ? So we can review easily :smile:

sined23, 2016-04-15:
Sorry, I don't know how to do it. You can review this new libraries and compare with existing
http://forum.arduino.cc/index.php?action=dlattach;topic=391757.0;attach=163027

sined23, 2016-04-15:
if you have questions for changes pls ask

sined23, 2016-04-15:
The main idea is that bytes are written and read into/from CDC buffer not one by one but block of bytes

sined23, 2016-04-15:
Serial_::accept write into CDC buffer as maximum as possible (to not overflow the buffer)
Serial_::readb (you can change just read, not use readb) get from CDC buffer as maximum as possible or nessasary length

sined23, 2016-04-15:
after that read speed is incredible increases !!!

NicoHood, 2016-04-15:
Fork the repo, do the changes, upload and open a pr. Its very simple, just learn some basic git commands (clone, add, commit, push).

sined23, 2016-04-15:
created PR

sined23, 2016-05-16:
Hi guys! Do you plan to merge this PR?
https://github.com/arduino/Arduino/pull/4871