arduino/Arduino #4869: 4 legs rgb led library for arduino

arsalmaner, 2016-04-15:
This library works for rgb led with 4 legs. You can manage colour rgb codes and work duration of the led light. 

[https://github.com/arsalmaner/Ardiuno-Libraries/tree/master/RGB_LED](url)

agdl, 2016-04-15:
Hi, your library cannot be added since a library.properties is missing. Please see how arduino library are structured and  follow [this guide](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ)

arsalmaner, 2016-04-15:
Thanks, I updated it with this guide. Can you check it out again?

agdl, 2016-04-15:
A tag is still missing

arsalmaner, 2016-04-18:
Sorry I am new. Is it ok now @agdl  ?

agdl, 2016-04-18:
Done thank you!

closed 2016-04-18 by agdl

per1234, 2016-04-29:
@arsalmaner your library was never added to Library Manager. The issue is the folder structure of your repository. From https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ:
>Your repo is ok if its root folder contains file library.properties and the rest of the library source code. If your library is stored into a subfolder, your repo it's not ok and we can't include your library in the list.

So you need to have a repository specifically for the library, rather than having the RGB_LED library be a subfolder of the Arduino-Libraries repository as it is currently.

arsalmaner, 2016-04-29:
I got it. Thanks @per1234 . I arranged it again. What should I do to request to add this now ?

agdl, 2016-05-03:
Argh.. @arsalmaner delete old tag and push it again. It will be added automatically!

arsalmaner, 2016-05-03:
ok @agdl I did it.

agdl, 2016-05-03:
Ok due to the way the manager works you will see changes on monday