arduino/Arduino #4870: New functionality in String class (WString.cpp/.h)

TobiasKnauss, 2016-04-15:
I need some new functions in the the String class.
I had created 2 posts about that in the arduino developers google group some hours ago, but the posts never appeared. Maybe the moderator is currently N/A...?

While waiting for the posts to appear, I have created the functionality and tested it.
It's ready to use and I'd be happy to see it implemented in the official version of the library:

```
WString.h:
  void trim(void);
  void trim(char remove);
  void trimStart(void);
  void trimStart(char remove);
  void trimEnd(void);
  void trimEnd(char remove);
  void padLeft(unsigned int newlength);
  void padLeft(unsigned int newlength, char add);
  void padRight(unsigned int newlength);
  void padRight(unsigned int newlength, char add);
  void cropLeft(unsigned int count);
  void cropRight(unsigned int count);
  void keepLeft(unsigned int count);
  void keepRight(unsigned int count);

WString.cpp:
void String::trim(void)
{
  trim (0x20);
}

void String::trim(char remove)
{
  trimStart(remove);
  trimEnd  (remove);
}

void String::trimStart(void)
{
  trimStart(0x20);
}

void String::trimStart(char remove)
{
  if (!buffer || len == 0 || remove == 0) return;
  char *begin = buffer;
  while ((*begin) == remove) begin++;
  len = buffer + len - begin;
  if (begin > buffer) memmove(buffer, begin, len);
  buffer[len] = 0;
}

void String::trimEnd(void)
{
  trimEnd(0x20);
}

void String::trimEnd(char remove)
{
  if (!buffer || len == 0 || remove == 0) return;
  char *end = buffer + len - 1;
  while ((*end) == remove && end >= buffer) end--;
  len = end + 1 - buffer;
  buffer[len] = 0;
}

void String::padLeft(unsigned int newlength)
{
  padLeft(newlength, 0x20);
}

void String::padLeft(unsigned int newlength, char add)
{
  if (newlength < len) newlength = len;
  if (!reserve(newlength)) return;
  unsigned int lenAdd = newlength - len;
  char *begin = buffer + lenAdd;
  memmove (begin, buffer, len);
  memset (buffer, add, lenAdd);
  len = newlength;
  buffer[len] = 0;
}

void String::padRight(unsigned int newlength)
{
  padRight(newlength, 0x20);
}

void String::padRight(unsigned int newlength, char add)
{
  if (newlength < len) newlength = len;
  if (!reserve(newlength)) return;
  unsigned int lenAdd = newlength - len;
  char *begin = buffer + len;
  memset (begin, add, lenAdd);
  len = newlength;
  buffer[len] = 0;
}

void String::cropLeft(unsigned int count)
{
  unsigned int lenKeep = len - count;
  keepRight(lenKeep);
}

void String::cropRight(unsigned int count)
{
  unsigned int lenKeep = len - count;
  keepLeft(lenKeep);
}

void String::keepLeft(unsigned int count)
{
  if (!buffer || len == 0) return;
  if (count <= len) len = count;
  buffer[len] = 0;
}

void String::keepRight(unsigned int count)
{
  if (!buffer || len == 0) return;
  if (count > len) count = len;
  char *begin = buffer + len - count;
  len = count;
  if (begin > buffer) memmove(buffer, begin, len);
  buffer[len] = 0;
}
```

TobiasKnauss, 2016-04-16:
```
I have created some more functionality: Every functions in the section 'modification' of WString.h has become 2 enhancements:

  // modification
  // notice:  void    function  (...) operates on           the object and returns void
  //          String  functionC (...) operates on a copy of the object and returns the copy (C=Copy)
  //          String* functionS (...) operates on           the object and returns a pointer to the object (S=Self)
  void    replace (char find, char replace);
  String  replaceC(char find, char replace);
  String* replaceS(char find, char replace);
  void    replace (const String& find, const String& replace);
  String  replaceC(const String& find, const String& replace);
  String* replaceS(const String& find, const String& replace);
  ...
  void    keepLeft (unsigned int count);
  String  keepLeftC(unsigned int count);
  String* keepLeftS(unsigned int count);
  void    keepRight (unsigned int count);
  String  keepRightC(unsigned int count);
  String* keepRightS(unsigned int count);
```

Again, tested and working.
cpp and h are attached.
My test program CString.ino is attached as well.

[WString.v2.zip](https://github.com/arduino/Arduino/files/222081/WString.v2.zip)
[CString.zip](https://github.com/arduino/Arduino/files/222082/CString.zip)

q2dg, 2016-04-16:
Thanks!!!

Meanwhile there is an "official" answer...could you pack these function into a library and upload it to Library Manager? I think this would give them more visibility to general users.

TobiasKnauss, 2016-04-17:
Hi,
do you mean this answer of yours? I cannot find any other "official" answer anywhere, and the posts in the google group are still missing.

I found an instruction list of how to upload a library:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ
but I think it doesn't make any sense in this case: I have enhanced the "String" class and kept it's name; I have NOT created a new one with different name. Therefore you'd have an ambigious reference if you added a library that holds a class with a name that already exists.
I uploaded the relevant files so that the Arduino developers can replace theirs by the new version. In this way the new functionality becomes a part of the Arduino "main libraries".

BTW: If you are trying to you the new files: On Windows there are 2 copies of WString.cpp/.h by default. You need to replace both.

Chris--A, 2016-04-18:
I'd recommend you fork the repo (and create a branch for your changes). Then you can create a pull-request to have the changes merged (and is easy for people to pull the proposed changes into a branch for testing before merge).

I guess we can link it to some other String class improvements I would like merged also.

https://github.com/arduino/Arduino/pull/2179
https://github.com/arduino/Arduino/pull/3096

TobiasKnauss, 2016-04-22:
With a delay of 5 days, the developer forum finally accepted and unlocked my post:
https://groups.google.com/a/arduino.cc/forum/#!topic/developers/ryrMAJiqwf0

@Chris--A : done as you suggested: https://github.com/TobiasKnauss/Arduino/
based on the master repo of this morning 2016-04-22 08:00 CEST