arduino/Arduino #4872: Way to change settings.path in installed IDE folder?

hanabanana, 2016-04-15:
I am using 1.6.7 IDE version. I wonder if there is a way to change the settings.path in the IDE installed folder instead of modifying the preferences.txt located at the %appdata% directory either manually or via CLI. I am asking because the IDE is installed on a network drive. I've downloaded the source for an additional board onto the same network drive. I want to set up the IDE such that the additional board is recognized by the IDE no matter which computer and which platform is trying to use this IDE. A solution to modify some "global preferences" text right in the IDE folder would be optimal for me.
 
I tried the preference.txt in /lib folder, but it didn't have any effect. I guess the local preferences has taken precedence.

matthijskooijman, 2016-04-16:
IIRC there is an option for "portable" installs, looks inside the install path for preferences.txt. I can't remember off-hand where it looks exactly, though (and no time right now to check).

per1234, 2016-04-16:
If you create a folder named portable in the Arduino IDE installation folder then the file {Arduino IDE install folder}/portable/preferences.txt will be used.

q2dg, 2016-04-16:
Is it documented somewhere? If it's not, it should (I suggest)

SbiriJJ, 2017-05-29:
https://www.arduino.cc/en/Guide/PortableIDE

closed 2017-07-04 by per1234