arduino/Arduino #4874: Modifying the bossac upload pattern in 1.6.6 to enable secure bit

steowens, 2016-04-16:
When I run the arduino installer code gets uploaded to my DUE but it uses the following flags:

``
/Users/%%%%/Library/Arduino15/packages/arduino/tools/bossac/1.6.1-arduino/bossac -i -d --port=cu.usbmodem1411 -U true -e -w -v -b /var/folders/7l/c6r4tz9d7mv7rrqz_dfyd610ydhv0l/T/build8aab6ec19f82baa2e91edda1b49f0ce1.tmp/HsmAes_V1.ino.bin -R 
``

But I cannot for the life of me find out how to modify this to set the -s flag as well.

matthijskooijman, 2016-04-16:
Try finding the hardware/arduino/sam/platform.txt file, that should define the commandline used for uploading. This file is probably somewhere in your users folder, I suspect it's /Users/%%%%/Library/Arduino15/packages/arduino/sam/platformt.txt.

agdl, 2016-07-25:
I think this can be closed. Please reopen this if I am wrong

closed 2016-07-25 by agdl