arduino/Arduino #4878: build.arch does not work

bobc, 2016-04-17:
Setting build.arch in platforms.txt does not work, it appears the builder substitutes the name of something else.

Workaround is to just ignore build.arch and hard code the name in compiler recipes e.g. 

-DARCH_KINETIS_K60

facchinm, 2016-04-18:
Hi @bobc , are you on IDE 1.6.8? From your description I can't understand what the real issue is, but _maybe_ it has been solved in master by merging https://github.com/arduino/arduino-builder/pull/125.
Could you test an hourly build and report if it works as intended? If it doesn't, could you expand the issue description (core used, action needed to reproduce the problem, full logs)? Thanks

bobc, 2016-04-19:
Hi @facchinm Yes, I'm using IDE 1.6.8 I tried nightly build and it has the same problem, plus an additional bug (missing include path)

I have attached files to illustrate the problem.

[platform.txt](https://github.com/arduino/Arduino/files/226797/platform.txt)

In platform.txt, at line 42, I set "build.arch=SAM".
At line 68, I pass the argument "-DARDUINO_ARCH_{build.arch} "

Therefore I would expect the expanded definition to read -DARDUINO_ARCH_SAM

[build_output.txt](https://github.com/arduino/Arduino/files/226802/build_output.txt)

In build_output.txt at line 17, the actual value is "-DARDUINO_ARCH_DUET"

It appears that build.arch is being set to the capitalized variant name or board name, not what I am setting.

Also attached, my boards.txt.

[boards.txt](https://github.com/arduino/Arduino/files/226810/boards.txt)

I hope that makes sense, thanks.

facchinm, 2016-04-20:
Hi @bobc ,
the builder populates that variable here https://github.com/arduino/arduino-builder/blob/master/src/arduino.cc/builder/setup_build_properties.go#L63.
To do it it uses the actual `PlatformId` (capitalized), dropping any override added in your platform.txt.
If you follow the instructions by @cmaglie on #4879 and rename your architecture to SAM (which is the correct way to inherit from another board)  it will automatically solve.

bobc, 2016-04-24:
I'm not inheriting from another board, I am creating new ones. "sam" is just an example name, I could have written "foo", or "bambleweeny57".

"Dropping an override" is the clue here. Overrides should drop the default case. If the default drops the override, it is not an override, is it?

I believe that settings in platform.txt should override whatever defaults are applied by Arduino builder, that is way all the other settings work, and not the other way round. Therefore I think you are confirming that there is a bug in the builder.

Even if you do not think it is a bug, please consider it as a feature request "a platform developer should be able to override default settings applied by the builder".