arduino/Arduino #4879: Builder runs preprocessor with wrong macro defines.

bobc, 2016-04-17:
When builder runs the preprocessor on the sketch for ctags, the compiler recipe is taken from global default, not the platform.txt file specific to the selected target.

So even if my platform.txt defines -DBOARD_DUET, the preprocessor gets called with -DARDUINO_DUET instead. The sketch subsequently does not parse correctly if it relies on BOARD_DUET being defined.

    C:\Programs\arduino-1.6.8\arduino-builder -dump-prefs -logger=machine -hardware "C:\Programs\arduino-1.6.8\hardware" -hardware "C:\Users\bob\AppData\Local\Arduino15\packages" -hardware "C:\Users\bob\Documents\Arduino\hardware" -tools "C:\Programs\arduino-1.6.8\tools-builder" -tools "C:\Programs\arduino-1.6.8\hardware\tools\avr" -tools "C:\Users\bob\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Programs\arduino-1.6.8\libraries" -libraries "C:\Users\bob\Documents\Arduino\libraries" -fqbn=Duet:Duet:duet -vid-pid=0X2341_0X003E -ide-version=10608 -build-path "C:\Users\bob\AppData\Local\Temp\build00ea9e9817149140aa6f35c88a9faceb.tmp" -warnings=default -prefs=build.warn_data_percentage=75 -verbose "C:\git_bobc\Marlin_bobc_RC\Marlin\Marlin.ino"
    C:\Programs\arduino-1.6.8\arduino-builder -compile -logger=machine -hardware "C:\Programs\arduino-1.6.8\hardware" -hardware "C:\Users\bob\AppData\Local\Arduino15\packages" -hardware "C:\Users\bob\Documents\Arduino\hardware" -tools "C:\Programs\arduino-1.6.8\tools-builder" -tools "C:\Programs\arduino-1.6.8\hardware\tools\avr" -tools "C:\Users\bob\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Programs\arduino-1.6.8\libraries" -libraries "C:\Users\bob\Documents\Arduino\libraries" -fqbn=Duet:Duet:duet -vid-pid=0X2341_0X003E -ide-version=10608 -build-path "C:\Users\bob\AppData\Local\Temp\build00ea9e9817149140aa6f35c88a9faceb.tmp" -warnings=default -prefs=build.warn_data_percentage=75 -verbose "C:\git_bobc\Marlin_bobc_RC\Marlin\Marlin.ino"
    Build options changed, rebuilding all
    "C:\Users\bob\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++"  -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -DF_CPU=84000000L -DARDUINO=10608 -DARDUINO_DUET -DARDUINO_ARCH_DUET  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Duet"'    "-IC:\Users\bob\Documents\Arduino\hardware\Duet\Duet\system/libsam" "-IC:\Users\bob\Documents\Arduino\hardware\Duet\Duet\system/libsam/include" "-IC:\Users\bob\Documents\Arduino\hardware\Duet\Duet\system/CMSIS/CMSIS/Include/" "-IC:\Users\bob\Documents\Arduino\hardware\Duet\Duet\system/CMSIS/Device/ATMEL/"   "-IC:\Users\bob\Documents\Arduino\hardware\Duet\Duet\cores\arduino" "-IC:\Users\bob\Documents\Arduino\hardware\Duet\Duet\variants\duet" "C:\Users\bob\AppData\Local\Temp\build00ea9e9817149140aa6f35c88a9faceb.tmp\sketch\Marlin.ino.cpp" -o "nul"
    "C:\Users\bob\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++"  -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -DF_CPU=84000000L -DARDUINO=10608 -DARDUINO_DUET -DARDUINO_ARCH_DUET  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Duet"'    "-IC:\Users\bob\Documents\Arduino\hardware\Duet\Duet\system/libsam" "-IC:\Users\bob\Documents\Arduino\hardware\Duet\Duet\system/libsam/include" "-IC:\Users\bob\Documents\Arduino\hardware\Duet\Duet\system/CMSIS/CMSIS/Include/" "-IC:\Users\bob\Documents\Arduino\hardware\Duet\Duet\system/CMSIS/Device/ATMEL/"   "-IC:\Users\bob\Documents\Arduino\hardware\Duet\Duet\cores\arduino" "-IC:\Users\bob\Documents\Arduino\hardware\Duet\Duet\variants\duet" "C:\Users\bob\AppData\Local\Temp\build00ea9e9817149140aa6f35c88a9faceb.tmp\sketch\Marlin.ino.cpp" -o "C:\Users\bob\AppData\Local\Temp\build00ea9e9817149140aa6f35c88a9faceb.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
    In file included from C:\Users\bob\AppData\Local\Temp\build00ea9e9817149140aa6f35c88a9faceb.tmp\sketch\thermistortables.h:26:0,
    
                     from C:\Users\bob\AppData\Local\Temp\build00ea9e9817149140aa6f35c88a9faceb.tmp\sketch\Configuration.h:943,
    
                     from C:\git_bobc\Marlin_bobc_RC\Marlin\Marlin.ino:37:
    
    C:\Users\bob\AppData\Local\Temp\build00ea9e9817149140aa6f35c88a9faceb.tmp\sketch\Marlin.h:42:26: fatal error: avr/pgmspace.h: No such file or directory
    
     #include <avr/pgmspace.h>
    
                              ^
    
    compilation terminated.
    
    exit status 1
    Error compiling for board Duet (Native USB Port).

matthijskooijman, 2016-04-18:
@bobc, from the top of my head, I think this means that your platform.txt does not define the recipe for the preprocessor, and arduino-builder falls back to a default. There have been some recent changes in this regard though, see https://github.com/arduino/arduino-builder/pull/125

cmaglie, 2016-04-19:
I see that your package is under:
`C:\Users\bob\Documents\Arduino\hardware\Duet\Duet\system/libsam`

but it should really be:
`C:\Users\bob\Documents\Arduino\hardware\Duet\sam\system/libsam`

would you like to try renaming `....\Duet\Duet` in `....\Duet\sam`?

Also if Duet is your board name, I suggest to rename the first folder to your "vendor/maintainer" name since the folders structure meaning is:
`hardware/<Vendor>/<Architecture>/....`

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#boardstxt

bobc, 2016-04-19:
Ok, I added the recipes for the preproc to my platform.txt, seems to work fine. I copied platform.txt from the latest Due package, but that works OK with the default recipes.

@cmaglie  thanks for the tip, I will change accordingly.

Just for general info, the Duet is a 3d printer board with onboard SAM3X8E, similar to Due (hence name), I am adapting the core package for use in Arduino IDE.

Anyway, I think this is solved, thanks.  I will close.

closed 2016-04-19 by bobc