arduino/Arduino #4880: Yet another 1.6.6+ (1.6.6 /1.6.7/1.6.8) issue involving UTFT library.

ghlawrence2000, 2016-04-18:
I am hoping somebody can get to the bottom of this since I can give you a repeatable situation to work with!

Henning Karlsen's UTFT libraries have been around for  a long time and worked flawlessly prior to IDE V 1.6.6. The situation I will describe is based on a Windows 7 x64 OS, with target board Arduino DUE.

Any of the example sketches, or indeed my own sketches using UTFT have the same problem under 1.6.6 or later.

The situation I have is this.... IDE V1.6.5 no problems at all, I can load a sketch (any sketch) which uses UTFT, compile it as many times as I wish without error.

IDE 1.6.6 onwards, I can load a sketch compile a single time without error, make NO changes, and try to compile again, I get an error.
`GetFileAttributesEx F:\Arduino\libraries\UTFT\UTFT.cpp F:\Arduino\libraries\UTFT\UTFT.h: The filename, directory name, or volume label syntax is incorrect.
Error compiling for board Arduino Due (Native USB Port).`

I have figured out how to get round it, but would like to know what is causing it.

With verbose during compilation enabled, I can locate the build folder, through trial and error, I have narrowed it down to \buildfolder\libraries\UTFT\ in this folder there are 4 files.
DefaultFonts.c.d
DefaultFonts.c.o
UTFT.cpp.d
UTFT.cpp.o

If I delete EITHER of the UTFT.cpp files..... compilation is successful, but if those files are both in the build folder, compilation WILL fail with the error previously stated.

Why specifically is UTFT causing this? I can have several Libraries in the build folder.... but the 'fix' remains the same, if I delete UTFT.cpp.d OR UTFT.cpp.O compilation is successful but will fail if BOTH of these files remain in the build folder.

I really hope you can do something about this as I cannot be the only one with this problem, and UTFT is certainly a common/popular library. I am using V2.81 downloaded directly from [rinkydinkelectronics](http://www.rinkydinkelectronics.com/library.php?id=51) If you require any further information please let me know.

Thanks,

Graham

matthijskooijman, 2016-04-18:
GetFileAttributesEx F:\Arduino\libraries\UTFT\UTFT.cpp F:\Arduino\libraries\UTFT\UTFT.h: The filename, directory name, or volume label syntax is incorrect.

There are two filenames in that error message. I suspect that the .d (dependencies) file is parsed incorrectly, and it ends up with two names in a single filename. Could you paste the full contents of the UTFT.cpp.d file?

ghlawrence2000, 2016-04-18:
Hi mattthijs,

Here is the entire file 
`C:\Users\Graham\AppData\Local\Temp\builde33ce9ddee6346054afe0349d71c85f0.tmp\libraries\UTFT\UTFT.cpp.o: \
 F:\Arduino\libraries\UTFT\UTFT.cpp F:\Arduino\libraries\UTFT\UTFT.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\cores\arduino/Arduino.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\cores\arduino/avr/pgmspace.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\cores\arduino/avr/interrupt.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\cores\arduino/binary.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\cores\arduino/itoa.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/libsam/chip.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/sam3xa.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/sam3x8e.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/CMSIS/Include/core_cm3.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/CMSIS/Include/core_cmInstr.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/CMSIS/Include/core_cmFunc.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/system_sam3xa.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_adc.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_can.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_chipid.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_dacc.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_dmac.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_efc.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_emac.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_gpbr.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_hsmci.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_matrix.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_pdc.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_pio.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_pmc.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_pwm.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_rstc.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_rtc.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_rtt.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_smc.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_spi.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_ssc.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_supc.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_tc.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_trng.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_twi.h \
 C:\Users\Graham\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.7\system/CMSIS/Device/ATMEL/sam3xa/include/component/component_uart.h \
[... cut ...]
 F:\Arduino\libraries\UTFT\hardware/arm/HW_ARM.h \
 F:\Arduino\libraries\UTFT\hardware/arm/HW_SAM3X8E.h \
 F:\Arduino\libraries\UTFT\memorysaver.h \
 F:\Arduino\libraries\UTFT\tft_drivers/ssd1289/initlcd.h \
 F:\Arduino\libraries\UTFT\tft_drivers/ssd1963/800/initlcd.h \
 F:\Arduino\libraries\UTFT\tft_drivers/cpld/initlcd.h \
 F:\Arduino\libraries\UTFT\tft_drivers/ssd1289/setxy.h \
 F:\Arduino\libraries\UTFT\tft_drivers/ssd1963/800/setxy.h \
 F:\Arduino\libraries\UTFT\tft_drivers/cpld/setxy.h
`
Thanks,

Graham

ghlawrence2000, 2016-04-26:
Frigging awesome....... after the initial SINGLE answer, nothing since..... Is this the best we can expect from the team??

:(

Graham

matthijskooijman, 2016-04-26:
I was actually working my way back through my mailbox after a busy period and looking into this just as you wrote that comment :-)

It seems arduino-builder improperly parses the .d file. There is enough info to fix this, just somebody has to sit down and do it now :-)

I've created https://github.com/arduino/arduino-builder/issues/136 to track this, since this is a bug in arduino-builder and not the Arduino IDE itself, so I'm closing this issue. If you have further comments, please leave them there.

closed 2016-04-26 by matthijskooijman