arduino/Arduino #4882: Wire library

Jexxio, 2016-04-18:
I'm trying to upload a sketch in order to use an DS3231 module, and i keep getting this error message

`Arduino: 1.6.8 (Mac OS X), Board: "Arduino/Genuino Uno"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/pierreugaz/Library/Arduino15/packages" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/pierreugaz/Library/Arduino15/packages" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/pierreugaz/Dropbox/Arduino/libraries" -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10608 -build-path "/var/folders/x2/vp_w26d153d31sc2gnggmvc40000gn/T/buildf5a22a99291579eaa19b528e7bcdfc7a.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/Users/pierreugaz/Dropbox/Arduino/libraries/DS3231/DS3231.ino"
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/pierreugaz/Library/Arduino15/packages" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/pierreugaz/Library/Arduino15/packages" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/pierreugaz/Dropbox/Arduino/libraries" -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10608 -build-path "/var/folders/x2/vp_w26d153d31sc2gnggmvc40000gn/T/buildf5a22a99291579eaa19b528e7bcdfc7a.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "/Users/pierreugaz/Dropbox/Arduino/libraries/DS3231/DS3231.ino"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/x2/vp_w26d153d31sc2gnggmvc40000gn/T/buildf5a22a99291579eaa19b528e7bcdfc7a.tmp/sketch/DS3231.ino.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src" "/var/folders/x2/vp_w26d153d31sc2gnggmvc40000gn/T/buildf5a22a99291579eaa19b528e7bcdfc7a.tmp/sketch/DS3231.ino.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src/Wire.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src/utility/twi.c" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src" "/var/folders/x2/vp_w26d153d31sc2gnggmvc40000gn/T/buildf5a22a99291579eaa19b528e7bcdfc7a.tmp/sketch/DS3231.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src" "/var/folders/x2/vp_w26d153d31sc2gnggmvc40000gn/T/buildf5a22a99291579eaa19b528e7bcdfc7a.tmp/sketch/DS3231.ino.cpp" -o "/dev/null"
Library can't use both 'src' and 'utility' folders. Double check /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire
Using library Wire at version 1.0 in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire 

Error compiling for board Arduino/Genuino Uno.`

harryboo, 2016-04-19:
Can u please check the latest nightly build, coz this should be resolved...

alvaromarlon, 2016-04-20:
Uso Arduino UNO, Win10 y la version arduino-1.6.8-windows.exe.
El problema es que no encuentra la librería "Wire.h" 

C:\Program Files (x86)\Arduino\libraries\LiquidCrystal_I2C\I2CIO.cpp:35:26: fatal error: ../Wire/Wire.h: No such file or directory

 #include <../Wire/Wire.h>

                          ^

compilation terminated.

exit status 1
Error compilación en tarjeta Arduino/Genuino Uno.

Necesito la librería y no se encuentra para descargarla.

harryboo, 2016-04-20:
@alvaromarlon:

this has nothing to do with the problem in this ticket.
As you can see, your include is made with an absolute path. This is the problem.

Please use one of these newer library versions:

https://github.com/fdebrabander/Arduino-LiquidCrystal-I2C-library
http://sparks.gogo.co.nz/LiquidCrystal_I2C_DFR.zip

facchinm, 2016-05-04:
@alvaromarlon , the solution proposed by @harryboo is correct, since the problem is a non-compliant library. I'm closing this issue as solved

closed 2016-05-04 by facchinm