arduino/Arduino #4886: Arduino IDE 1.6.8 Blocks access to all serial ports

dlabun, 2016-04-22:
On Windows it appears that IDE 1.6.8 is claiming access to every single serial port on the PC. This is really undesirable as it prevents you from using the serial ports for any other tasks.

rglenn, 2016-05-01:
Noticed this as well, haven't had time to dig into it too much yet. It comes and goes - I'll try 5 times to upload to an ESP8266 or connect to a serial port with TeraTerm, and it won't work, but the 6th or so will.

Thinking maybe it's scanning the available serial ports, and opening them to do it? Need to investigate more.

rglenn, 2016-05-01:
@dlabun This might be a the same issue as #4700

dlabun, 2016-05-01:
Could very well be the same issue, it's hard to tell with the length of the conversation around #4700. In my opinion for "advanced" users there should be a preference to disable continuous scanning of the serial ports and only scan on upload.

rglenn, 2016-05-01:
Related pull request #4792 links to http://downloads.arduino.cc/liblistSerials/liblistSerials-1.1.0.zip as the new replacement for Arduino\lib\liblistSerialsj.dll - that seems to have resolved it for me.

facchinm, 2016-05-02:
The fix for this problem has been merged in mainline so it will be included in 1.6.9.
In the meantime you could use the nightly version as it already contains the fix :smile:

closed 2016-05-02 by facchinm