arduino/Arduino #4888: Arduino IDE 1.6.8 takes incredibly long to open

Goz3rr, 2016-04-22:
Splash screen gets stuck on "Loading configuration..." for several minutes.
Got this from booting with `-Xprof`

```
Loading configuration...

Flat profile of 1.05 secs (65 total ticks): AWT-Shutdown

  Interpreted + native   Method
100.0%     1  +     0    sun.awt.AWTAutoShutdown.run
100.0%     1  +     0    Total interpreted

  Thread-local ticks:
 98.5%    64             Blocked (of total)

Initializing packages...
Preparing boards...
Starting...

Flat profile of 0.99 secs (58 total ticks): AWT-Shutdown

  Interpreted + native   Method
100.0%     0  +     1    java.security.AccessController.getStackAccessControlContext
100.0%     0  +     1    Total interpreted

  Thread-local ticks:
 98.3%    57             Blocked (of total)

Flat profile of 0.99 secs (58 total ticks): AWT-EventQueue-0

  Thread-local ticks:
100.0%    58             Blocked (of total)

Flat profile of 243.34 secs (15065 total ticks): main

  Interpreted + native   Method
 98.7%     0  + 14871    java.io.FileOutputStream.close0
  0.1%    10  +     6    java.lang.ClassLoader.defineClass1
  0.1%     0  +    14    java.io.WinNTFileSystem.getBooleanAttributes
  0.1%     0  +    12    java.io.FileInputStream.open0
  0.1%     0  +     8    java.net.NetworkInterface.getAll
  0.0%     0  +     4    java.io.WinNTFileSystem.list
  0.0%     0  +     2    java.util.zip.Inflater.inflateBytes
  0.0%     2  +     0    javax.swing.UIDefaults.getUI
  0.0%     0  +     2    sun.nio.fs.WindowsNativeDispatcher.CreateFile0
  0.0%     0  +     2    java.lang.ClassLoader.findBootstrapClass
  0.0%     0  +     2    sun.awt.windows.ThemeReader.getPartSize
  0.0%     0  +     2    java.security.AccessController.doPrivileged
  0.0%     2  +     0    org.apache.batik.dom.util.HashTable.<init>
  0.0%     1  +     0    java.util.zip.ZipFile.getEntrySize
  0.0%     0  +     1    java.net.NetworkInterface.getMacAddr0
  0.0%     1  +     0    java.lang.Class.getDeclaredFields0
  0.0%     0  +     1    java.io.WinNTFileSystem.canonicalizeWithPrefix0
  0.0%     0  +     1    sun.java2d.loops.GraphicsPrimitiveMgr.registerNativeLoops
  0.0%     0  +     1    java.util.zip.Inflater.reset
  0.0%     0  +     1    sun.awt.windows.WComponentPeer.addNativeDropTarget
  0.0%     0  +     1    sun.awt.windows.WWindowPeer.reshapeFrame
  0.0%     0  +     1    sun.misc.Unsafe.defineAnonymousClass
  0.0%     1  +     0    java.lang.Class.forName0
  0.0%     0  +     1    sun.nio.fs.WindowsNativeDispatcher.FindNextFile
  0.0%     0  +     1    java.lang.Float.floatToRawIntBits
 99.6%    56  + 14943    Total interpreted (including elided)

     Compiled + native   Method
  0.0%     0  +     1    java.lang.Object.<init>
  0.0%     0  +     1    java.net.URL.<init>
  0.0%     0  +     2    Total compiled

         Stub + native   Method
  0.2%     0  +    31    java.io.WinNTFileSystem.getBooleanAttributes
  0.0%     0  +     5    java.lang.Object.getClass
  0.0%     0  +     3    java.lang.Thread.currentThread
  0.0%     0  +     3    java.lang.System.arraycopy
  0.0%     0  +     2    java.util.zip.ZipFile.getEntry
  0.0%     0  +     2    java.lang.Object.clone
  0.0%     0  +     1    java.lang.System.nanoTime
  0.0%     0  +     1    java.lang.ClassLoader.findBootstrapClass
  0.0%     1  +     0    java.security.AccessController.doPrivileged
  0.0%     0  +     1    sun.misc.Unsafe.getInt
  0.0%     0  +     1    java.lang.Class.isPrimitive
  0.0%     0  +     1    java.lang.Object.hashCode
  0.0%     0  +     1    java.lang.String.intern
  0.0%     0  +     1    java.util.zip.Inflater.inflateBytes
  0.4%     1  +    53    Total stub

  Thread-local ticks:
  0.0%     5             Blocked (of total)
  0.0%     5             Class loader

Flat profile of 0.26 secs (7 total ticks): Thread-19

  Thread-local ticks:
100.0%     7             Blocked (of total)

Flat profile of 0.73 secs (34 total ticks): Thread-20

  Interpreted + native   Method
 70.6%     0  +    24    java.net.SocketInputStream.socketRead0
 14.7%     0  +     5    java.net.DualStackPlainSocketImpl.connect0
  5.9%     0  +     2    java.net.Inet6AddressImpl.lookupAllHostAddr
  2.9%     1  +     0    sun.net.www.http.KeepAliveStream.<clinit>
  2.9%     1  +     0    java.net.URI$Parser.parseHierarchical
  2.9%     0  +     1    java.io.FileInputStream.open0
100.0%     2  +    32    Total interpreted

Flat profile of 5.01 secs (290 total ticks): Image Fetcher 0

  Thread-local ticks:
100.0%   290             Blocked (of total)

Flat profile of 6.02 secs (349 total ticks): Image Fetcher 3

  Interpreted + native   Method
100.0%     0  +     1    java.lang.Object.wait
100.0%     0  +     1    Total interpreted

  Thread-local ticks:
 99.7%   348             Blocked (of total)

Flat profile of 5.00 secs (299 total ticks): Keep-Alive-Timer

  Thread-local ticks:
100.0%   299             Blocked (of total)
```

dlabun, 2016-04-23:
I don't have any issues on 1.6.8... What version of Java are you using?

Goz3rr, 2016-04-23:
```
>java -version
java version "1.8.0_60"
Java(TM) SE Runtime Environment (build 1.8.0_60-b27)
Java HotSpot(TM) Client VM (build 25.60-b23, mixed mode)
```

dlabun, 2016-04-23:
My computer with Win7 just updated to 1.8.0-77 yesterday... Maybe try updating?

Goz3rr, 2016-04-23:
I'm pretty sure the IDE comes bundled with 1.8.0_66 so updating wouldn't do anything

dlabun, 2016-04-24:
Even the bundled version still searches for updates unless you disabled it... The bundled version still installs like full version but doesn't add the shortcuts or control panel link. Check your Java bin folder for javacpl.exe and from there you'll be able to search for available updates and access the diagnostics info.

Of course I could totally wrong about this being Java related...

agdl, 2016-06-22:
@Goz3rr please try the last arduino nightly build and let us know if performances are improved. Have you any anti-virus installed? If so maybe the access scan stacks the IDE startup

Goz3rr, 2016-06-22:
Apologies, I have since reinstalled Windows after which it was fixed and totally forgot about this issue

closed 2016-06-22 by Goz3rr