arduino/Arduino #4889: Unbeleivable

KLEINTREE, 2016-04-23:
Recently got into Arduino and spent months learning c++structure....But am seriously beside myself as at every juncture there is complete and utter failure! web site links do not work, referneces to files or folders that do not exist, hell ten minutes ago i added #include <firmata.h> just below my other header call and now have a compile error (this program was running minutes before with this being the only added entry)..Even trying to contact arduino...every link is broken!!! WTF are these guys really this incomeptent? I am an electrical contractor and do alot of process control  and was hoping form something new to use but after some seroius consideration this Arduino is about as stable a system as the wind! Hell the simple fact that almost none of the blog information is hopeful for real problems just screams "stay away!!! we sell it but dont want to talk to you abou YOUR problems"". I have experienced almost every single line error on the Support pages and not one of the solutions on this site helped!!!! This is funny, but the first thing I did was try to install blink (factory made example program) with a compile error (after almost walking away just from problems trying to get communication established
I gotta say, after  delving into this for some time now I have to agree with those that say tArduino is not the way to go in almost any capacity, I could not even plug in and communicate without hours of downloads, reading blogs(literally thousand of dead ends and REALLY BAD information ) always starting at Arduino-never any help except to point you to another dead end, deleting, re-installing, and finally coming up with a solution on some Processing programmers web site that enabled me to move forward..this took about 12 hours total and just be aware that I do complex industrial automation logic structures but not too in depth with programming working some in C++ recently, Basic, Pascal, Lisp, diesel,...Aftert all this and at every single step running into the simplest fixes with no support is way to heavy a burden, better off starting from scratch in C++ (or maybe even any other language for that matter) because you will have a far better chance of having a stable operating system.If you really want to make things work, I would not go this route because it will soon fade away and become obsolete. they can only sell dysfunctional system for a while until people get wind of the failure and go elsewhere. I do understand the simplicity "theory" here and that is why I spent so much time in Arduino, but the utter failure at EVERY juncture is really beyond comprehension. And these problems are with the simples I/O operations. Now, as the program expands and gets more complex, this turns into a freakin black hole of complications. If you are at ll serious -even as a hobbysit- dont waste you efforts learning Arduino as it is possiby the most problematic language i have ever seen (take a look at the bugs report--then try to get something to work-EVEN THE STANDARD EXAMPLES). Also understand that EVERY programming language has bugs and even strange anomalies that occur when particular (or even random) events happen and it is normal to have to resolve a few things like that on any platform, but holy cow you  will not even get to that stage with this  language. I do no know or understand what they did to c++, but I am done reasearching Ardunio as its kinda like polishing a turd, if the foundation isnt good, the entire structure WILL fail.

dlabun, 2016-04-23:
Which Arduino website are you talking about? I have not seen any website issues related to dead links.

PaulStoffregen, 2016-04-23:
Is this trolling?

If it really is a genuine complaint, of missing info & broken links & incomplete info, which is of course completely absent of any specific links & references to any specific cases is pretty astounding hypocrisy.

KLEINTREE, 2016-04-24:
Your comment is expected. Ignorant, apathetic,and gonna stay that way. No
ear for a customer just a little bad mouth. Good Dy Mr Arduino, this should
good posted too, you have no concern for cusotmer concerns whatsoever.
Prior Purchase Cusotmer

KLEINTREE, 2016-04-24:
Here You Go! No attempt to rectify, remedy, or even INQUIRE about a
customer complaint JUst jump right in and bad  mouth--i had to test it
because that is what i am seeing on their website and othe blogs are
pointing that way.. No doubt, you can get Arduino working ( i have done it)
but holy cow the hours and problems spent in the process are
exhaustable....I can plug just about any PLC in and be programming within
1/2 hour...The proof is in the pudding, see above for how these guys will
treat you. There are many Arduino "spawns" that will provide at least some
support, and maybe even try to help when you make a complaint.

dlabun, 2016-04-24:
Yeah, just a non-English speaking troll....

ncguk, 2016-04-24:
@KLEINTREE Can you supply an example or two of the broken links you're experiencing? I'm not affiliated with Arduino, but I'm sure those who are would be willing to look into this.

mdilger, 2016-04-26:
The same guy posted crap on the openHAB forum yesterday.  Definitely a troll....

cmaglie, 2016-04-26:
Closing this issue as invalid.

closed 2016-04-26 by cmaglie