arduino/Arduino #4893: grbl library for arduino nano

ehsan1200, 2016-04-25:
hi every one,i wanna run cnc by aurduino nano and now i need its library,i found uno library but ineed nano
thanks

matthijskooijman, 2016-04-25:
@ehsan1200, this is something you should take up with the grbl maintainers. This repository is meant for problems and suggestions in the Arduino environment itself, so we cannot really help you here (though from what I know of Grbl, I'm not sure is a nano version exists. You could probably run grbl on a nano too, but you would have to dig into the specifics of the pinout and pin-to-port mapping to figure out where to connect things, and perhaps change some code to match the nano).

closed 2016-04-25 by matthijskooijman

ehsan1200, 2016-04-26:
@matthijskooijman so thanks