arduino/Arduino #4900: program again after unplugged

camohe90, 2016-04-26:
In the latest version 1.6.8 each time that i unplugged the arduino from the USB port and I plug again i need to program again.

Paul-Ver, 2016-05-03:
Had/have the same issue. I think it's because the way Arduino works in general (resets when you open a serial connection).

When I upload a program. It'll start running (in which I use hardware serial).
But when I reconnect the Arduino, it'll not run on default, it will only run when I open the Serial Monitor.

facchinm, 2016-05-03:
@camohe90 , could you expand your issue? Like adding the board you are using, operating system, sketch you are trying to run, if it works on 1.6.7, if it work in nighlty

@Paul-Ver , I don't believe the problem is related with `!Serial()`, otherwise it shouldn't work after programming it (unless you open the serial monitor)

camohe90, 2016-05-03:
@facchinm i was using an arduino uno, the operating system is windows 10. I try with different sketch. And  the only way to solve the problem is uninstall 1.6.8 and install again the 1.6.7 and everything worka perfect again

facchinm, 2016-05-03:
@camohe90 greta, thanks for the info. Probably you are encountering a side effect of https://github.com/arduino/Arduino/issues/4700 which we already fixed in mainline.
Could you test the nightly version of the IDE and report if it works? Thanks

agdl, 2016-07-25:
I'm closing this for lack of feedback. Please reopen it if needed.

closed 2016-07-25 by agdl