arduino/Arduino #4906: ArduinoISP Windows 10 driver could not installed

DieterIllingen, 2016-04-27:
Hi,
I downloaded the actual driver for ArduinoISP but Windows 10 tells me that the driver could not installed.
I followed getting started, but it will not reflect Windows 10. Is there a new driver? 

Thanks
   Dieter

agdl, 2016-06-22:
@DieterIllingen I also had problems in installing these drivers on windows 10. However this product is a discontinued one in the arduino product line. Try using the adafruit installer
https://github.com/adafruit/Adafruit_Windows_Drivers/releases/download/1.0.0.0/adafruit_drivers.exe
The only difference will be that instead of selecting ArduinoISP as programmer you will have to choose USBTiny

facchinm, 2016-07-19:
@agdl workaround works correctly, closing as solved

closed 2016-07-19 by facchinm