arduino/Arduino #4913: Library Manager add Matrix

tomstewart89, 2016-04-30:
I've written a library to handle matrices and matrix operations on Arduino. Would it be possible to add it to the library manager? The github repo can be found here:

https://github.com/tomstewart89/Matrix

Thanks!

PaulStoffregen, 2016-05-01:
In the first several years, Arduino shipped with 2 libs named Matrix & Sprite.  They were removed at some point...

Naming a new library "Matrix" might lead to some confusion?

tomstewart89, 2016-05-02:
That's a fair point. It seems like by and large "matrix" is used to refer to LED matrices, so a different name is probably a good idea. In that case, why don't I call it BasicLinearAlgebra instead?

https://github.com/tomstewart89/BasicLinearAlgebra

agdl, 2016-05-03:
Done thank you!

closed 2016-05-03 by agdl