arduino/Arduino #4919: Arduino 101 concatenating float/double to string as wrong value for -1 to 0

sanelss, 2016-05-03:
When adding a float/double type between the values of -1 to 0 it returns it as a positive number.. example code:

void setup() {
  Serial.begin(115200);
}

void loop() {
  String s="";
  float db=-0.5;
  //db+=1.1; //doing this will make the string print positive 0.6 so the value is correct
  s+=db;  //when adding double/float type to string you get a positive number for values of -1 to 0
  Serial.println(s);
  delay(100);
}

facchinm, 2016-05-03:
Hi @sanelss ,
the issue had been fixed in mainline with this commit (https://github.com/01org/corelibs-arduino101/commit/7bd93b3baf802481282e32240fcfc1f941e36561), it will be published soon as `Curie core 1.0.6`.
Closing as already fixed

closed 2016-05-03 by facchinm

sanelss, 2016-05-03:
Thank you much for the quick reply. I have manually modified the file to add fix though i think this should have been published immediately once caught as I see it as a major bug

sanelss, 2016-05-03:
Upon further investigation when doing this modification by itself I still get the same behavior as before. this does not appear to be a fix for the issue

facchinm, 2016-05-03:
Sorry, maybe I linked you the wrong commit, however if you apply all the post 1.0.5 patches it works (tested :smile: )

facchinm, 2016-05-03:
And Intel has almost monthly release schedule, so it won't take longer before it gets released

sanelss, 2016-05-03:
Just a headsup, pulling the latest build does indeed fix the issue(though you did say you tested it by still) :D so yea the commit you posted isn't enough/right part but it works now.

facchinm, 2016-05-03:
Ok, the right commit was https://github.com/01org/corelibs-arduino101/commit/a54f5132630fb2825e27d84cf4ddf3d8ee22192a , thanks for reporting !

lancelon, 2017-05-01:
Hi Sanel please can you contact me? Thanks