arduino/Arduino #4920: EMoRo 2560 Librbary

inovatic-ict, 2016-05-03:
Hello,

company Inovatic ICT is Arduino's partner and we would like that user can download libraries for EMoRo 2560 via Arduino Manager.

Link to json: 

https://raw.githubusercontent.com/inovatic-ict/inovatic-repoz/master/package_Inovatic-EMoRo_index.json

Thanks!

Dino

agdl, 2016-05-16:
Done thank you!

P.S next time link to the repo please :)

closed 2016-05-16 by agdl

inovatic-ict, 2016-05-16:
OK! Thanks!

When will these libraries be able to download via Library Manager?

agdl, 2016-05-16:
It should be already available. After the adding in about one hour it should be there

inovatic-ict, 2016-05-16:
So why I can't see it?

Sketch->Include Library-> Manage Libraries -> typing Emoro and nothing shows up.

And is it possible to download our Emoro2560 Bord files via the Bord Manager without using the json file, ? We want that our users do not need to put json URL in the „Preferences“, than just types „Emoro2560“ in the „Bord Manager“ and install Board files and libreries.

Json URL: https://raw.githubusercontent.com/inovatic-ict/inovatic-repoz/master/package_Inovatic-EMoRo_index.json

Thank you!