arduino/Arduino #4923: Serial Monitor laggy if not scrolling

NicoHood, 2016-05-04:
I use debug prints in my code. And I noticed once the scroll wheel is not exceeded, the serial monitor printing lags. Once the bottom is reached and the monitor has to scroll upwards for new text it is very fluent again.

This is very annoying as I am trying to debug time critical code. I have to first fill the whole screen to debug my code.

Using IDE 1.6.9 from git, linux x64, Arduino Leonardo.

PaulStoffregen, 2016-05-04:
I've seen this too on Ubuntu 14.04.  It's happening on the Linux side, where the window manager appears to be restricting the update rate of windows without keyboard focus.

matthijskooijman, 2019-10-01:
I believe some Serial monitor performance improvements have been made since 2016. @NicoHood could you test if this problem still exists (or if not, close this issue)?