arduino/Arduino #4925: Serial Monitor on Win10

giacomopetrussa, 2016-05-04:
Hi, 
I'm using the serial monitor for my project.
One week ago, I was using the Arduino compiler on my Win10 and I ran the Serial Monitor... it didn't appear! I explain it better.. the monitor was in the taskbar, but I could see it ONLY from the Aero animation, only. I tried 3 different versions of the compiler. Always the same problem.
If you have got any solution, please, contact me.
Thank you so much.

PaulStoffregen, 2016-05-04:
Do you use 2 or more monitors, perhaps only sometimes?

Maybe Windows is configured for a large multi-monitor desktop, and perhaps the serial monitor is appearing elsewhere in the space that's not shown on your single monitor?

giacomopetrussa, 2016-05-04:
Good evening Paul. Thanks for your interest.
I've always used one monitor...I rebooted the PC several times. Always the
same problem: I can't see my serial monitor. I also tried some older
versions of Arduino compiler, but nothing.. I remember that two days before
the problem I shut down the PC by removing the battery (you know Windows
and his f...ing problems.. Haha) And I think that it was doing something,
but it can't be the cause, I think. Probably there's a incompatibility
thanks to the new updates?

carlosperate, 2016-05-12:
If you shift + right click on the task bar icon you should be able to see a menu with the "move" option. As soon as you click that option the mouse will go to the title-bar of the selected window, **do not** do anything with the mouse at this point! instead press any of the arrow keys in your keyboard, this will activate the "move action", and at this point it will be "latched to your mouse", so if you move the mouse around now (do not do any click, just move it around) you should see the window latched to your cursor.

See if that gives you an idea of what could be happening to the window.

agdl, 2016-06-22:
@giacomopetrussa sometimes happen (It happened to me) that the serial monitor window becomes very very very narrow and so you can't even see it. You simply have to enlarge it. Let me know if this help!

agdl, 2016-06-22:
I'm closing this. Reopen if needed

closed 2016-06-22 by agdl