arduino/Arduino #4926: Switching monitors breaks IDE GUI

NicoHood, 2016-05-04:
If you plug in a new monitor on linux or change the monitor settings the gui will get a lot of errors and sometimes you need to fully quit it to recovery any usable gui element. It happens always. Using elementary os, based on ubuntu 14.04, intel graphics (on board, laptop, display port happens with all 3 monitors).

```
Exception in thread "AWT-EventQueue-0" java.lang.ClassCastException: sun.awt.image.BufImgSurfaceData cannot be cast to sun.java2d.xr.XRSurfaceData
	at sun.java2d.xr.XRPMBlitLoops.cacheToTmpSurface(XRPMBlitLoops.java:145)
	at sun.java2d.xr.XrSwToPMBlit.Blit(XRPMBlitLoops.java:353)
	at sun.java2d.pipe.DrawImage.blitSurfaceData(DrawImage.java:959)
	at sun.java2d.pipe.DrawImage.renderImageCopy(DrawImage.java:577)
	at sun.java2d.pipe.DrawImage.copyImage(DrawImage.java:67)
	at sun.java2d.pipe.DrawImage.copyImage(DrawImage.java:1014)
	at sun.java2d.pipe.ValidatePipe.copyImage(ValidatePipe.java:186)
	at sun.java2d.SunGraphics2D.drawImage(SunGraphics2D.java:3318)
	at sun.java2d.SunGraphics2D.drawImage(SunGraphics2D.java:3296)
	at processing.app.EditorStatus.paintComponent(EditorStatus.java:232)
	at javax.swing.JComponent.paint(JComponent.java:1056)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JSplitPane.paintChildren(JSplitPane.java:1047)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JLayeredPane.paint(JLayeredPane.java:586)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paintToOffscreen(JComponent.java:5217)
	at javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1579)
	at javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1502)
	at javax.swing.BufferStrategyPaintManager.paint(BufferStrategyPaintManager.java:306)
	at javax.swing.RepaintManager.paint(RepaintManager.java:1272)
	at javax.swing.JComponent.paint(JComponent.java:1042)
	at java.awt.GraphicsCallback$PaintCallback.run(GraphicsCallback.java:39)
	at sun.awt.SunGraphicsCallback.runOneComponent(SunGraphicsCallback.java:79)
	at sun.awt.SunGraphicsCallback.runComponents(SunGraphicsCallback.java:116)
	at java.awt.Container.paint(Container.java:1975)
	at java.awt.Window.paint(Window.java:3912)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:842)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:814)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:814)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:789)
	at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:738)
	at javax.swing.RepaintManager.access$1200(RepaintManager.java:64)
	at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1732)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Exception in thread "AWT-EventQueue-0" java.lang.ClassCastException: sun.awt.image.BufImgSurfaceData cannot be cast to sun.java2d.xr.XRSurfaceData
	at sun.java2d.xr.XRPMBlitLoops.cacheToTmpSurface(XRPMBlitLoops.java:145)
	at sun.java2d.xr.XrSwToPMBlit.Blit(XRPMBlitLoops.java:353)
	at sun.java2d.pipe.DrawImage.blitSurfaceData(DrawImage.java:959)
	at sun.java2d.pipe.DrawImage.renderImageCopy(DrawImage.java:577)
	at sun.java2d.pipe.DrawImage.copyImage(DrawImage.java:67)
[... 214 lines cut ...]
	at java.awt.Window.paint(Window.java:3912)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:842)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:814)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:814)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:789)
	at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:738)
	at javax.swing.RepaintManager.access$1200(RepaintManager.java:64)
	at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1732)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

leogaggl, 2016-10-04:
Can confirm this issue also on Ubuntu 16.04 LTS with a dual-monitor setup. When system goes on standby and comes back after the switching of monitors with the following errors in the log console and the Editor is just white. 

Exception in thread "AWT-EventQueue-0" java.lang.ClassCastException: sun.awt.image.BufImgSurfaceData cannot be cast to sun.java2d.xr.XRSurfaceData
	at sun.java2d.xr.XRPMBlitLoops.cacheToTmpSurface(XRPMBlitLoops.java:145)
	at sun.java2d.xr.XrSwToPMBlit.Blit(XRPMBlitLoops.java:353)
	at sun.java2d.SurfaceDataProxy.updateSurfaceData(SurfaceDataProxy.java:498)
	at sun.java2d.SurfaceDataProxy.replaceData(SurfaceDataProxy.java:455)
	at sun.java2d.SurfaceData.getSourceSurfaceData(SurfaceData.java:233)
	at sun.java2d.pipe.DrawImage.renderImageCopy(DrawImage.java:566)
	at sun.java2d.pipe.DrawImage.copyImage(DrawImage.java:67)
	at sun.java2d.pipe.DrawImage.copyImage(DrawImage.java:1014)
	at sun.java2d.SunGraphics2D.drawImage(SunGraphics2D.java:3318)
	at sun.java2d.SunGraphics2D.drawImage(SunGraphics2D.java:3296)
	at com.sun.java.swing.plaf.gtk.GTKEngine.paintCachedImage(GTKEngine.java:555)
	at com.sun.java.swing.plaf.gtk.GTKPainter.paintScrollBarThumbBackground(GTKPainter.java:1133)
	at javax.swing.plaf.synth.SynthScrollBarUI.paintThumb(SynthScrollBarUI.java:322)
	at javax.swing.plaf.synth.SynthScrollBarUI.paint(SynthScrollBarUI.java:279)
	at javax.swing.plaf.synth.SynthScrollBarUI.update(SynthScrollBarUI.java:244)
	at javax.swing.JComponent.paintComponent(JComponent.java:780)
	at javax.swing.JComponent.paint(JComponent.java:1056)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JComponent.paintToOffscreen(JComponent.java:5210)
	at javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1579)
	at javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1502)
	at javax.swing.BufferStrategyPaintManager.paint(BufferStrategyPaintManager.java:306)
	at javax.swing.RepaintManager.paint(RepaintManager.java:1272)
	at javax.swing.JComponent._paintImmediately(JComponent.java:5158)
	at javax.swing.JComponent.paintImmediately(JComponent.java:4969)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:831)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:814)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:814)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:789)
	at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:738)
	at javax.swing.RepaintManager.access$1200(RepaintManager.java:64)
	at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1732)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

cmaglie, 2016-10-04:
Looks like a bug in the JDK:

https://stackoverflow.com/questions/34188495/how-can-i-work-around-the-classcastexception-in-java2d-bug-id-7172749

NicoHood, 2016-10-04:
But this bug is marked as resolved. I just unplugged my Monitor and got the same lengthy error again. I am using arch so I have the latest java software.

cmaglie, 2016-10-04:
what's the output of your `java -version`?
the bug seems fixed in jdk8 update 112

NicoHood, 2016-10-04:
```
[arch@arch avr]$ java -version
openjdk version "1.8.0_102"
OpenJDK Runtime Environment (build 1.8.0_102-b14)
OpenJDK 64-Bit Server VM (build 25.102-b14, mixed mode)
```

pabloab, 2017-05-10:
Might be related to bug #3501. Same problem on Ubuntu 16.04, Arduino 18.2: 
```
java -version
openjdk version "1.8.0_121"
OpenJDK Runtime Environment (build 1.8.0_121-8u121-b13-0ubuntu1.16.04.2-b13)
OpenJDK 64-Bit Server VM (build 25.121-b13, mixed mode)
```

d-a-v, 2018-09-25:
Same problem when my laptop is waking up from lid closed:
Arduino 1.8.7, openjdk version "10.0.2" 2018-07-17 64bits, Ubuntu 18.04.1 LTS
`arduino Exception in thread "AWT-EventQueue-0" java.lang.ClassCastException: sun.awt.image.BufImgSurfaceData cannot be cast to sun.java2d.xr.XRSurfaceData`