arduino/Arduino #4927: [Library Manager] Request for adding Infrared4Arduino to library manager

bengtmartensson, 2016-05-05:
This is a request to add [Infrared4Arduino](https://github.com/bengtmartensson/Infrared4Arduino) to the library manager. Thank you.

NicoHood, 2016-05-05:
Thanks for posting! Also have a look at my implementation which is quite different. Maybe you can find something useful. Currently the dev2 branch is the most up to date version, but still not complete. For sending IR data I still have no good idea yet.
https://github.com/NicoHood/IRLremote/tree/dev2

agdl, 2016-05-06:
Done thank you!

closed 2016-05-06 by agdl

bengtmartensson, 2016-05-06:
@agdl: Thank you!

@NicoHood: Thanks for your comment. Actually, I am aware of your library since some time. Your work put the emphasis on the efficiency, in my work, cleanliness and structure (in the object oriented sense) has priority. So there is definitely room for more than one!