arduino/Arduino #4929: Arduino Library not working with Edison board

jabreo, 2016-05-06:
Hi - I'm using the Intel Edison board with the Grove Arduino shield on an sketch to drive the dual I2C motor controllers.  When I use the Intel Arduino 101 board the code works fine.  When I use the Intel Edison board nothing happens.  I've contacted Intel and they say to work with Arduino.cc because you maintain the Arduino IDE libraries.  Interestingly enough, the blink sketch works fine on the Edison board.

Right now I'm using:
- Arduino IDE version 1.6.7 on Mac OS 10.10.5
- Intel i686 Boards library version 1.6.7+1.0
- Intel Curie Boards library version 1.0.5 (when using the Arduino 101 board)

When I use the Arduino 101 board, I’m using the "Intel Curie (32 bit) Boards – Arduino/Genuino 101” board library

When I use the Edison board, I’m using the “Arduino i686 Boards – Intel® Edison” board library
I’m also making sure I use the correct serial port –  /dev/cu.usbmodemxxxx

I appreciate your help in looking into this.

-Jarred

Here is the code I'm using:
------------------------------------------------------------------------------------------------------
#include <Wire.h>

#define MotorSpeedSet         0x82
#define MotorSetA                 0xa1
#define MotorSetB                 0xa5
#define Nothing                     0x01
#define DirectionSet              0xaa
#define I2CMotorDriverAdd         0x0f   // Set the address of the I2CMotorDriver

void setup()  {
  Wire.begin(); // join i2c bus (address optional for master)
  delayMicroseconds(1000); //wait for motor driver to initialization
}

void MotorSpeedSetAB(unsigned char MotorSpeedA , unsigned char MotorSpeedB)  {
  MotorSpeedA=map(MotorSpeedA,0,100,0,255);
  MotorSpeedB=map(MotorSpeedB,0,100,0,255);
  Wire.beginTransmission(I2CMotorDriverAdd); // transmit to device I2CMotorDriverAdd
  Wire.write(MotorSpeedSet);                 // set pwm header 
  Wire.write(MotorSpeedA);                   // send pwma 
  Wire.write(MotorSpeedB);                   // send pwmb    
  Wire.endTransmission();                     // stop transmitting
}

//set the direction of DC motor. 
void MotorDirectionSet(unsigned char Direction)  {     //  Adjust the direction of the motors 0b0000 I4 I3 I2 I1
  Wire.beginTransmission(I2CMotorDriverAdd); // transmit to device I2CMotorDriverAdd
  Wire.write(DirectionSet);                  // Direction control header
  Wire.write(Direction);                     // send direction control information
  Wire.write(Nothing);                       // need to send this byte as the third byte(no meaning)  
  Wire.endTransmission();                    // stop transmitting 
}
 
void loop()  
{
    MotorSpeedSetAB(100,20);
    delay(10); //this delay needed
    MotorDirectionSet(0b1010);  //0b1010  Rotating in the positive direction 
    delay(1000); 
    MotorDirectionSet(0b0101);  //0b0101  Rotating in the opposite direction
    delay(500);
}

sandeepmistry, 2016-05-27:
Hi @SidLeung @calvinatintel,

What's the best place to report this to get resolved? I think it's most likely something in the Edison's Wire library which is bundled with the core.

However, this issue tracker does not seem to be enabled in the Github repo for the core: https://github.com/01org/corelibs-edison

facchinm, 2016-07-19:
@calvinatintel @eriknyquist  maybe this issue can be solved/analysed in a future revision of i686 core

eriknyquist, 2016-07-19:
@facchinm @sandeepmistry , Jarred has since contacted me through other channels. Currently waiting for motor controllers (which have been out of stock for a while) to be delivered, and working with @jabreo to get enough info to recreate the issue here.

sandeepmistry, 2016-07-19:
@eriknyquist thanks for the update! Since this is being tracked in another system and is not linked to the IDE, or AVR/SAM cores I will close this for now.

Please let us know where to redirect Edison related issues in the future.

closed 2016-07-19 by sandeepmistry