arduino/Arduino #4933: Atmel QTouch Library in Arduino IDE?

sabas1080, 2016-05-09:
Hi

can work with Atmel QTouch Library in Arduino IDE? with the introduction of ATMEL328PB 
It would be interesting

Thanks

syddo, 2016-05-12:
This is a good idea, I'm watching this thread. :)

mbanzi, 2016-05-12:
Arduino doesn't support pre-compiled libraries so supporting libraries like QTouch  has always been an issue.

syddo, 2016-05-13:
okay... I never looked at the QTouch Libraries so I just learned about that.

touch is very popular and I still believe that it is a good idea for Arduino to support it.
is there a group looking into this now?

NicoHood, 2016-05-16:
There is already another post about using precompiled binaries like .o and .a files. It would not be too complicated to add this feature though.

syddo, 2016-06-25:
I agree @NicoHood ...
PLus I beleive Atmel now will release new QTouch enabled MCUs pretty soon.

per1234, 2017-07-04:
The feature request for pre-compiled libraries support: https://github.com/arduino/Arduino/issues/4336

per1234, 2017-11-03:
The Arduino IDE now supports .a and .so files in libraries. See instructions in the `precompiled` property documentation here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

So can we close this issue now?

kgoveas, 2017-11-17:
@per1234 ,

Can you give an example of a library.properties file, name of the static lib, and how the folder structure should look like? It doesn't seem to recognize a .a file.

per1234, 2017-11-21:
I wrote the documentation for the `precompiled` property and my previous comment based on @facchinm's comment on the Arduino Developers mailing list:
https://groups.google.com/a/arduino.cc/d/msg/developers/YH0CNIednmw/ZvmMcN5kAQAJ
That post indicated that support for the `precompiled` property had been added in arduino-builder 1.3.25. However, after looking at the release history I now see that https://github.com/arduino/arduino-builder/pull/219 was merged after the 1.3.25 release and there has been no arduino-builder release since.

It appears from your other comment: https://github.com/arduino/arduino-builder/pull/219#issuecomment-345255702 that you also realized that @kgoveas.

So I think it would be best to remove the documentation for the `precompiled` property until the next release.

If anyone does verify that the QTouch library can be used with the Arduino IDE please do comment here so we can close this issue.

kgoveas, 2017-11-21:
Hey @per1234 ,

Yes indeed. But one can still use precompiled libs with a few hacks to the Arduino IDE 1.8.5. 
1) Replace the arduino-builder with the [test build](http://downloads.arduino.cc/PR/arduino-builder/arduino-builder-219.zip) from @facchinm  
2) Change the position of the compiler.c.elf.extra_flags in the platform.txt file to the end of the other linker flags. I've reported an issue for the [sam](https://github.com/arduino/ArduinoCore-sam/issues/46) and [samd](https://github.com/arduino/ArduinoCore-samd/issues/278) cores.

P.S. I have updated the [documentation](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#library-metadata) a bit, but we need to mention that it is currently not supported? It would help people such as myself finish making the libraries in advance to the next Arduino IDE release.

per1234, 2017-11-21:
Thanks for your work on the wiki @kgoveas! I have updated it to correct the version information but left the documentation in place. It's difficult to know how to best handle documentation for new features that have not yet been released (or even merged). Ideally documentation would be added in the same commit as the code changes that require it but that's not possible with the wiki. This means documentation updates end up being done as an afterthought or neglected altogether. Other than that issue I do like the use of the GitHub Wiki for Arduino's developer level documentation.

per1234, 2020-09-29:
Arduino's precompiled library support is now fairly mature and well documented. If anyone is interested, please see the Arduino Library Specification:
https://arduino.github.io/arduino-cli/latest/library-specification/#precompiled-binaries

I'm going to assume that it is now possible to use the QTouch library and close this.

closed 2020-09-29 by per1234