arduino/Arduino #4939: not declared in this scope

zenmanenergy, 2016-05-10:
I just upgraded from 1.6.5 to 1.6.8 and some of my code broke because of the tabs.

If you have two tabs and a function is defined in the right tab and called in the left tab it throws an error: 

'functionName' is not declared in this scope

But if I simply copy and paste the same function code above where it is called then the error goes away. This didn't used to happen in previous versions of the IDE. 

I recognize using .h files to declare the functions will fix this, but that's a more complicated process for a beginner. This error makes tabs sort of useless.

facchinm, 2016-05-10:
Hi @zenmanenergy ,
the problem you are experiencing is similar to the one being tracked in https://github.com/arduino/arduino-builder/issues/140 ; we merged a lot of fixes between 1.6.8 and the just released 1.6.9, so if you want to give it a spin and see if the problem is fixed for you it would be great!
Please report the results on the `arduino-builder` issue, thanks!

closed 2016-05-10 by facchinm

zenmanenergy, 2016-05-10:
I just installed and tested it. This is not fixed in v 1.6.9

facchinm, 2016-05-10:
Thanks for testing. Could you also test the builder linked here? https://github.com/arduino/arduino-builder/pull/131#issuecomment-216813118 

If also this doesn't work please comment on arduino/arduino-builder#140 with your PC specifications (OS version), the board you are targeting, core version of this board (you can get it from board manager) and an example sketch triggering the problem, thanks!

zenmanenergy, 2016-05-10:
Just tested it on a MacOSX El Cap 10.11.4 with Arduino 1.6.9 IDE and the downloaded arduino-builder file in the /Contents/Java folder and it gives me the exact same error.

zenmanenergy, 2016-05-10:
This is using an esp8266 01 board using the v2.2.0 core arduino codebase.

facchinm, 2016-05-10:
Using esp core 2.2.0 and IDE 1.6.9 a simple sketch with 
```
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:
  foo();
}
```
and on the other tab
```
void foo() {}
```

works correctly. I believe that your issue has already been opened somewhere, so please attach the sketch (or a minimal sketch that triggers the issue).

zenmanenergy, 2016-05-10:
Scratch that, I was testing the wrong file! I had two files I was working with that had the same name, I apologize.

It IS fixed. Keep this closed. Thank you facchinm

zenmanenergy, 2016-05-10:
Or rather... it is fixed with the 1.3.1 version of the builder. It's still broken in current building in v1.6.9

facchinm, 2016-05-10:
Good to know :smile: Thanks for reporting! The PR containing the fix should be merged in the next few days!

zenmanenergy, 2016-05-10:
Thanks again. And thanks for working on the IDE. It's frustrating at times when things break, but overall it's awesome!