arduino/Arduino #4942: I typed a function that prevent screen tearing on Arduino

meokullu, 2016-05-10:
Hello fellas.

As I mentioned on subject I typed very simple function that prevent screen from tearing. People faces with this problem when they want to print some data on screen without delay and/or when they have to clear screen on loop.

Example:

1.  
void loop()
{
lcd.print(analogRead(A0));
}
It will look like vibrations on characters of screen.

They have to use delay() function that provided by Arduino for this problem but with single processor, as you know, delay() function kills functionality of Arduino when we use delay().

with the function that I typed, they will not use any delay, so they will gain some time for do other things or do this faster as it can be.

How can I contact with Arduino engineers for discuss my function about it and add it to arduino for all users.

PS: I am a unemployed engineer of computer science from Turkey who enjoys arduino.
for contact if it is necessary: meokullu@gmail.com and my name is Enes.

agdl, 2016-06-06:
@meokullu the best way you can propose tour change is to submit a pull request to the LCD library and if valid it will be merged for sure.

agdl, 2016-06-07:
You can submit this here https://github.com/arduino-libraries/LiquidCrystal

I'm closing this issue

closed 2016-06-07 by agdl