arduino/Arduino #4943: Arduino my void loop stops after 1 cycle

liandranth, 2016-05-11:
Hi, I've recently started programming Arduino sketches.
I'm tring to make a controller of temperature, read current consumption of a pc and a monitor, moreover I need to read the rpm of two set of fans.
I only knew how to acquire temperature and use that information to control the fans, for read the current value I've copied an already written program that use a specific library to run; the program is called Energy Monitor and it's based on the Emonlib, at last for read the fan rpm I've copied a program that incorporates the attach interrupts.
Now when i run the program, on the serial monitor are printed all the values that i need but after the first cycle it doesn't print anything.
I'm asking for some help to understad why it stops, i post the code below, if you have some suggestion about how to acquire the current consumption without using a library based program I'll be grateful (I'm using a LEM HXS-20NP hall effect sensor).
I'm thinking that the Energy monitor program generates some problems in the regular flow of the program.

Here there is the link from which I copied the Energy monitor program:
https://openenergymonitor.org/emon/buildingblocks/how-to-build-an-arduino-energy-monitor

I've uploaded the library files of the Energy Monitor program that i used on dropbox:
https://www.dropbox.com/s/iy8cwrqoku5n27m/EmonLibmodded.zip?dl=0

Here it starts the code:

`#include "EmonLib.h"
EnergyMonitor emon1;
EnergyMonitor emon2;

const byte errorPin=13;
int ris=6, ven=10, ven2=11, door=8,ledDoor=12;
word val;
float temperatura, temp;
int NbTopsFan,NbTopsFan1,calc,calc1, hallsensor=2,sta;
byte libero=5, libero1=9;                                    //unused pins that i'll need afterwards
volatile byte state=LOW;

typedef struct{                                            //Defines the structure for multiple fans and their dividers     
  char fantype;
  unsigned int fandiv;
}fanspec;                                                   //Definitions of the fans
fanspec fanspace[3]={{0,1},{1,2},{2,8}};
char fan=1;

void activate()
{
  NbTopsFan++;
}

void activate1()
{
  NbTopsFan1++;
}

void setup() {
Serial.begin(9600);

pinMode(errorPin,OUTPUT);
pinMode(hallsensor,INPUT);
pinMode(3,INPUT);
pinMode(door,INPUT);
pinMode(ledDoor,OUTPUT);
pinMode(ris,OUTPUT);
pinMode(ven,OUTPUT);
attachInterrupt(0,activate,RISING);
attachInterrupt(1,activate1,RISING);

emon1.voltage(A2,234.26,1.7);
emon1.current(A1,111.1);

emon2.voltage(A4,234.26,1.7);
emon2.current(A3,111.1);

}

void loop() {

Serial.flush();

emon1.calcVI(20,2000);
Serial.println("PC_Sensor:");
emon1.serialprint();
emon1.statusPC();

emon2.calcVI(20,2000);
Serial.println("Monitor_Sensor:");
emon2.serialprint();
emon2.statusMonitor();

val=analogRead(A0);
temperatura=(float)val/1023*5;
temp=(float)temperatura/0.01-50;

Serial.print("temperatura: ");
Serial.print(temp);
Serial.println("^C"); 

NbTopsFan = 0;  
   sei();   //Enables interrupts
   delay (1000);  
   cli();   //Disable interrupts
   calc = ((NbTopsFan * 60)/fanspace[fan].fandiv);       

//Times NbTopsFan (which is apprioxiamately the fequency the fan is spinning at) by 60 seconds //before dividing by the fan's divider                                                                                                                                                                     
                                                                                                                                              
   Serial.print (" set ventole 1: ");
   Serial.print (calc, DEC);
   Serial.print (" rpm\r\n");

NbTopsFan1 = 0;  
   sei();   //Enables interrupts
   delay (1000);  
   cli();   //Disable interrupts
   calc1 = ((NbTopsFan1 * 60)/fanspace[fan].fandiv);                                                                                    
   
   Serial.print (" set ventole 2: ");
   Serial.print (calc1, DEC);
   Serial.print (" rpm\r\n");

sta=digitalRead(door);
digitalWrite(ledDoor,sta);

digitalWrite(errorPin,state);

 if(val<143){
    digitalWrite(ris,HIGH);                             //controllo temperatura interna
    digitalWrite(ven,LOW);
    digitalWrite(ven2,LOW);}                        //superata la soglia avvia le ventole
  if(143<=val<=174){
    digitalWrite(ris,LOW);
    digitalWrite(ven,LOW);
    digitalWrite(ven2,LOW);}
  if(val>174){
    digitalWrite(ris,LOW);
    digitalWrite(ven,HIGH);
    digitalWrite(ven2,HIGH);}
  
  if(digitalRead(ven)==HIGH && NbTopsFan==0){                         //controlla se supera la temperatura
    digitalWrite(errorPin,HIGH);                                                     //e se le ventole entrano in funzione
    Serial.println("errore: il set di ventole 1 non sta girando");}     //in caso non funzionasse segnala l'errore
    if(digitalRead(ven)==HIGH&&NbTopsFan!=0){
      digitalWrite(errorPin,LOW);}
   
   if(digitalRead(ven)==HIGH && NbTopsFan1==0){                           
    digitalWrite(errorPin,HIGH);                                         
    Serial.println("errore: il set di ventole 2 non sta girando");}     
    if(digitalRead(ven)==HIGH&&NbTopsFan1!=0){
      digitalWrite(errorPin,LOW);}

}`

Thank you for any suggestion.

liandranth, 2016-05-11:
I'm now sure that my problem is related to the serialprint() function, defined in the EmonLib.cpp file that i've uploaded.

Chris--A, 2016-05-11:
This is a question for the Arduino forum:

http://forum.arduino.cc/index.php?board=4.0

This is an issue tracker for the IDE and its related software's. Specific coding problems relating to your sketch belong on the forum. :)

liandranth, 2016-05-11:
ok thank you, I'll post this issue to the appropriate forum.

matthijskooijman, 2016-05-11:
What @Chris--A said, indeed. I'm closing this issue.

closed 2016-05-11 by matthijskooijman